xbianonpi/xbian #805: Xbian-update upgrade stopping I2S sound output.

coderpop, 2016-01-25:
Every time I upgrade xbian-update package since the New Year, I2S audio output ceases after the reboot. The last xbian-update package to not do this is 20151221 (from memory).

Even tried again yesterday with the same problem. I2S tested and working, xbian-config, upgrade xbian-config, reboot, no sound...

The I2S hardware is detected and correctly reported with aplay -l and alsa is selected in kodi and the DT settings are in /boot/config.txt.....But yet there is no sound!

mkreisl, 2016-01-25:
Only one idea I have to figure out what's your problem is:

When sounds does not work, what says `lsmod | grep lirc`? If the output is something like this
```
lirc_rpi                6106  0 
lirc_dev                7794  1 lirc_rpi
rc_core                15912  1 lirc_dev
```
LIRC is probably blocking I2S and then look to `dmesg | grep lirc`
```
[   14.418455] lirc_dev: IR Remote Control driver registered, major 244 
[   14.436624] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[   14.439763] lirc_rpi: to_irq 188
[   15.395112] lirc_rpi: auto-detected active high receiver on GPIO pin 22
[   15.398158] platform lirc_rpi: lirc_dev: driver lirc_rpi registered at minor = 0
[   15.398187] lirc_rpi: driver registered!
```
In my configuration I'm using a Hifiberry Amp+ (I2S) and LIRC at the same time, but I have remapped the LIRC input pin (using GPIO 22):

```
...
# GPIO pins have to be remapped for LIRC for working together with Hifiberry
dtoverlay=lirc-rpi,gpio_out_pin=16,gpio_in_pin=22,debug

# Hifiberry
dtoverlay=hifiberry-amp
...
```

Just did all updates from stable and staging - sound is still fine :smile:

coderpop, 2016-02-21:
Sorry for the delay in replying, my Father has been ill.

No that is not the answer to the problem. I tried again today and with everything updated to todays latest with the exception of xbian-update and all is well. Do the upgrade to xbian-update and it stops the sound.

lsmod | grep lirc has nothing either before or after the upgrade. Alsa-base and alsa-utils are still installed and the latest. Kodi sound output still has Alsa selected for output. In /boot/config.txt the DT lirc overlay is still commented out.

lsmod | grep dac gives the following....
xbian@cinema ~ $ lsmod | grep dac
snd_soc_hifiberry_dac     2154  0
snd_soc_core          120296  3 snd_soc_hifiberry_dac,snd_soc_pcm5102a,snd_soc_bcm2708_i2s
snd_pcm                67774  3 snd_soc_core,snd_soc_hifiberry_dac,snd_pcm_dmaengine
xbian@cinema ~ $

So.....The mystery continues! ;)

mkreisl, 2016-03-08:
@coderpop 
And the module `snd_bcm2835` is definitely NOT loaded? 
See also here http://forum.xbian.org/thread-2397-post-30868.html#pid30868

coderpop, 2016-03-10:
The installation is working until xbian-update is upgraded, then it breaks.... So unless xbian-update loads snd_bcm2835 then I don't think that is the issue. 

Maybe I will get some time at the weekend to try it out again.

coderpop, 2016-04-06:
Posted on forum as well, but not everyone sees that, so here's an update....

It still happens. It seems that updating 'xbian-update' overrides settings in /etc/modprobe.d/xbian.conf and /etc/modules,xbian - Both files are somehow changed to add back snd_bcm2835 during or after the update.
Manually changing the files back to how they were and the sound returns!

closed 2016-06-17 by mkreisl