xbianonpi/xbian #806: XBian often does not boot after a reboot

Smultie, 2016-01-25:
My XBian install (XBian staging/devel/stable) often does not boot the first time after a reboot.
What steps can I take to find the cause of this failure to boot.

After unplugging and replugging the power connector, it almost always works again.

mkreisl, 2016-01-25:
@Smultie 
Does the system not properly shutdown or boot?

Maybe you don't wait enough :smile: Sometimes my RPi1 neets up to 3mins until shutdown will be finished

A view into `/var/log/upstart-ev.log` may be useful 

And applying this patch will also log oldstyle SysV processes (got it from @mk01):
```
--- a/etc/init.d/rc
+++ b/etc/init.d/rc
@@ -123,7 +123,9 @@
 			shift
 			scripts="$@"
 			for script in $scripts ; do
+initctl emit -n starting JOB="$script"
 				$debug "$script" $action
+initctl emit -n stopped JOB="$script"
 			done
 		}
 		;;
```

Smultie, 2016-01-25:
> Maybe you don't wait enough [image: :smile:] Sometimes my RPi1 neets up

I gave it a "sudo reboot" command through SSH (and VPN)
The Pi never came back online ;)

Guess when I'm back home and (re)plug the power cable, all is fine.

Will look into upstart-ev.log to see if I see anything out of the ordinary.

Thanks for your help thus far!

Smultie, 2016-01-25:
Anything you can deduct from below?

Mon Jan 25 13:26:16 CET 2016 65042.69 258154.72 stopped mountall                         1.32 0.43 0.26 3/346 27799
Mon Jan 25 13:27:11 CET 2016 65097.73 258366.87 started reboot-force                     0.80 0.47 0.29 6/356 28015
Mon Jan 25 13:27:11 CET 2016 65097.73 258366.87 runlevel                                 0.80 0.47 0.29 4/355 28015
Mon Jan 25 13:27:11 CET 2016 65097.95 258366.95 stopped avahi-daemon                     0.80 0.47 0.29 15/307 28083
Mon Jan 25 13:27:11 CET 2016 65097.95 258366.95 stopped cron                             0.80 0.47 0.29 10/298 28085
Mon Jan 25 13:27:11 CET 2016 65097.97 258366.95 stopped tty2                             0.80 0.47 0.29 13/288 28088
Mon Jan 25 13:27:11 CET 2016 65097.96 258366.95 stopped upstart-file-bridge              0.80 0.47 0.29 14/295 28087
Mon Jan 25 13:27:11 CET 2016 65097.98 258366.95 stopped usbmount-cleaner                 0.80 0.47 0.29 8/280 28091
Mon Jan 25 13:27:11 CET 2016 65097.98 258366.95 stopped udev                             0.80 0.47 0.29 8/276 28091
Mon Jan 25 13:27:11 CET 2016 65097.97 258366.95 stopped xbian-inotifyd-bridge            0.80 0.47 0.29 7/269 28092
Mon Jan 25 13:27:11 CET 2016 65098.00 258366.95 stopped ntp                              0.80 0.47 0.29 4/260 28093
Mon Jan 25 13:27:11 CET 2016 65098.20 258367.08 stopped idmapd                           0.80 0.47 0.29 12/279 28149
Mon Jan 25 13:27:11 CET 2016 65098.20 258367.08 stopped nmbd                             0.80 0.47 0.29 12/276 28149
Mon Jan 25 13:27:12 CET 2016 65098.20 258367.08 stopped lirc                             0.80 0.47 0.29 10/277 28146
Mon Jan 25 13:27:12 CET 2016 65098.20 258367.08 stopped dbus                             0.80 0.47 0.29 15/273 28149
Mon Jan 25 13:27:12 CET 2016 65098.22 258367.08 stopped hwclock.sh                       0.80 0.47 0.29 7/264 28153
Mon Jan 25 13:27:12 CET 2016 65098.20 258367.08 stopped tty1                             0.80 0.47 0.29 5/260 28153
Mon Jan 25 13:27:12 CET 2016 65098.21 258367.08 stopped upstart-udev-bridge              0.80 0.47 0.29 7/260 28154
Mon Jan 25 13:27:12 CET 2016 65098.23 258367.08 stopped frandom                          0.80 0.47 0.29 4/253 28154
Mon Jan 25 13:27:12 CET 2016 65098.25 258367.09 stopped autofs                           0.80 0.47 0.29 4/248 28166
Mon Jan 25 13:27:12 CET 2016 65098.26 258367.09 stopped zram-swap                        0.80 0.47 0.29 3/244 28166
Mon Jan 25 13:27:12 CET 2016 65098.32 258367.17 started wait-for-state  xbmc             0.80 0.47 0.29 4/242 28175
Mon Jan 25 13:27:12 CET 2016 65098.72 258367.79 stopped xbmc-screensaver                 0.80 0.47 0.29 7/241 28202
Mon Jan 25 13:27:12 CET 2016 65098.76 258367.81 started xbian-xbmc-cpufreq               0.80 0.47 0.29 6/239 28210
Mon Jan 25 13:27:12 CET 2016 65098.84 258367.87 started xbmc-priority                    0.80 0.47 0.29 5/246 28233
Mon Jan 25 13:27:12 CET 2016 65099.08 258368.12 stopped xbmc-priority                    0.80 0.47 0.29 4/238 28251
Mon Jan 25 13:27:13 CET 2016 65099.46 258368.90 stopped sickrage                         0.80 0.47 0.29 2/181 28267
Mon Jan 25 13:27:17 CET 2016 65103.61 258384.11 stopping xbmc                            0.74 0.46 0.29 1/140 28292
Mon Jan 25 13:27:17 CET 2016 65103.74 258384.42 starting xbmc-exits                      0.74 0.46 0.29 6/155 28327
Mon Jan 25 13:27:17 CET 2016 65103.74 258384.42 started xbmc-changesettings              0.74 0.46 0.29 7/154 28328
Mon Jan 25 13:27:17 CET 2016 65103.75 258384.42 stopped xbmc                             0.74 0.46 0.29 5/148 28328
Mon Jan 25 13:27:17 CET 2016 65103.82 258384.45 started rc                               0.74 0.46 0.29 6/149 28349
Mon Jan 25 13:27:17 CET 2016 65103.82 258384.45 stopped xbian-xbmc-cpufreq               0.74 0.46 0.29 6/147 28357
Mon Jan 25 13:27:17 CET 2016 65103.85 258384.45 stopped wait-for-state  xbmc             0.74 0.46 0.29 3/142 28358
Mon Jan 25 13:27:17 CET 2016 65103.87 258384.46 stopped openbsd-inetd                    0.74 0.46 0.29 3/139 28366
Mon Jan 25 13:27:17 CET 2016 65103.91 258384.49 stopped xbmc-changesettings              0.74 0.46 0.29 6/141 28382
Mon Jan 25 13:27:17 CET 2016 65103.95 258384.53 started wait-for-state  xbmc-exits       0.74 0.46 0.29 3/144 28399
Mon Jan 25 13:27:17 CET 2016 65103.98 258384.56 stopped tvonoff-xbmc                     0.74 0.46 0.29 4/143 28408
Mon Jan 25 13:27:17 CET 2016 65104.02 258384.61 stopped xbmc-exits                       0.74 0.46 0.29 5/137 28422
Mon Jan 25 13:27:17 CET 2016 65104.05 258384.64 stopped wait-for-state  xbmc-exits       0.74 0.46 0.29 4/139 28439
Mon Jan 25 13:27:18 CET 2016 65104.59 258385.70 started progress-shutdown                0.74 0.46 0.29 2/136 28533
Mon Jan 25 13:27:19 CET 2016 65105.90 258389.31 stopped progress-shutdown                0.74 0.46 0.29 3/131 28578
Mon Jan 25 13:27:19 CET 2016 65106.15 258389.98 splash                                   0.76 0.47 0.29 5/134 28613
Mon Jan 25 13:27:20 CET 2016 65106.49 258390.80 unmounted-remote-filesystems             0.76 0.47 0.29 2/135 28662
Mon Jan 25 13:27:20 CET 2016 65106.51 258390.83 deconfiguring-networking                 0.76 0.47 0.29 1/135 28669
Mon Jan 25 13:27:20 CET 2016 65107.11 258392.57 stopped networking                       0.76 0.47 0.29 3/133 28757
Thu Jan  1 01:00:03 CET 1970 3.36 9.60 starting udev                            0.00 0.00 0.00 8/129 235
[... cut ...]
Mon Jan 25 13:27:36 CET 2016 29.32 30.26 stopped mounted-tmp                      2.24 0.55 0.18 13/285 2278
Mon Jan 25 13:27:36 CET 2016 29.38 30.26 stopped wait-for-state  xbmc-done        2.24 0.55 0.18 12/281 2281
Mon Jan 25 13:27:37 CET 2016 29.42 30.26 stopped screen-cleanup                   2.24 0.55 0.18 7/276 2283
Mon Jan 25 13:27:37 CET 2016 29.80 30.26 stopped xbmc-failed-start                2.24 0.55 0.18 6/278 2300
Mon Jan 25 13:27:39 CET 2016 31.97 30.71 started zram-swap                        2.54 0.64 0.22 4/271 2320

mkreisl, 2016-01-25:
For me it looks fine
So I would assume the network connection does not work properly
(the mounts in /mnt, are they locally (usb) or network mounts?)

Smultie, 2016-01-25:
Network mounts.
Could that be the culprit?

It's a NAS that does hibernate, so maybe the mounts weren't reachable?

mkreisl, 2016-01-25:
Hmmm, but the log says /mnt/movies is ok. And, the NAS should wake up automatically, isn't it?

Smultie, 2016-01-25:
Yeah, you're right. Any other logs that might give something away?

coderpop, 2016-01-25:
I have experienced the same issue, but only on Rpi1, not on Rpi2.

I guess you have an Rpi1 Smultie?

Smultie, 2016-01-25:
Nope, RPi2 :)

oxivanisher, 2016-01-26:
It seems, that I have the same issue. RPi 2, fresh installed and no network or USB mounts. It "starts" kod, but then it exists without showing the GUI. The kodi log file (debug activated) is very unhelpful why it is exiting. SSH works fine.

mk01, 2016-02-12:
@Smultie 

apply the commit or edit the file, after deconfiguring network is pretty much done - what comes is just remount rootfs RO and reboot(shutdown).

this we will see only after the edit as @mkreisl told you. then the output will look more like this:
```
Fri Feb 12 01:13:45 CET 2016 13984.20 50289.76 stopped /etc/rc6.d/K04rsyslog            0.50 0.32 0.29 2/130 13444
Fri Feb 12 01:13:45 CET 2016 13984.30 50290.04 stopped /etc/rc6.d/K05umountnfs.sh       0.50 0.32 0.29 2/126 13473
Fri Feb 12 01:13:46 CET 2016 13984.56 50290.97 unmounted-remote-filesystems             0.50 0.32 0.29 3/127 13492
Fri Feb 12 01:13:46 CET 2016 13984.58 50291.00 deconfiguring-networking                 0.50 0.32 0.29 4/127 13499
Fri Feb 12 01:13:46 CET 2016 13984.80 50291.61 stopped networking                       0.50 0.32 0.29 3/130 13572
Fri Feb 12 01:13:46 CET 2016 13984.80 50291.61 stopped /etc/rc6.d/K06umountfs           0.50 0.32 0.29 3/125 13573
Fri Feb 12 01:13:46 CET 2016 13985.27 50293.01 stopped /etc/rc6.d/K07umountroot         0.50 0.32 0.29 3/124 13699
Fri Feb 12 01:13:46 CET 2016 13985.39 50293.31 stopped /etc/rc6.d/K08reboot             0.50 0.32 0.29 2/124 13726
```

Smultie, 2016-02-12:
Would love to do that @mk01, but I really don't know how!

mk01, 2016-02-12:
copy this
```
--- a/etc/init.d/rc
+++ b/etc/init.d/rc
@@ -123,7 +123,9 @@
            shift
            scripts="$@"
            for script in $scripts ; do
+initctl emit -n starting JOB="$script"
                $debug "$script" $action
+initctl emit -n stopped JOB="$script"
            done
        }
        ;;
```

ssh into rpi, do:
```
sudo -s
cd /
vi mini.patch[hit enter]
```

then press "i" paste into window. push ESC, release it. put ":w" hit enter, push ESC, release it. put ":q". you should be back in command prompt.
now:
```
patch -p1  < mini.patch
```

should be it

mk01, 2016-02-12:
also, if you change "splash" into "nosplash" in /boot/cmdline.txt , on start and reboot/shutdown you will see that console log on screen ... (to have idea what is going on)

Smultie, 2016-02-12:
Hunk #1 FAILED at 123.
1 out of 1 hunk FAILED -- saving rejects to file etc/init.d/rc.rej

Used nano iso vi, but I assume that's all the same for this purpose?!

mk01, 2016-02-12:
@Smultie 

apparently the change is contained in pool of commits to xbian-update I had in queue (and just pushed).
so in max ~4h there will be xbian-update, just install it (in staging).

Smultie, 2016-02-12:
Thanks, xbian-update installed. Still need to run the script, or was that
part of the update?

mkreisl, 2016-02-12:
@Smultie I would assume: NO/YES :smile:

NO --> running script not needed
YES --> is part of the update

This is now in:
```
                        action=$1
                        [ $action = stop ] && uaction=stopped || uaction=started
                        shift
                        scripts="$@"
                        for script in $scripts ; do
                                initctl emit -n $uaction JOB="$script"
                                $debug "$script" $action
                        done
```

closed 2023-04-29 by mkreisl