xbianonpi/xbian #807: Playback get stuck at the specific point on RPI v2

Stunherald, 2016-01-31:
Hello all, 

I'm having issues with one of the video files. 
Video playback is OK for several minutes. Suddenly audio stops to play and also subtitles get stuck, video continues to play. After few seconds video will also get stuck.

It is happening at the exact same spot every time, so it's not random thing. Also notice that the debug overlay info in top-left corner is changing, so it is not a signal lost or something like that.

Debug log from kodi with component specific enabled for SMB, FFmpeg and Video component - [videoissue.zip](https://github.com/xbianonpi/xbian/files/111319/videoissue.zip)

Video i took showing what is happening on TV- [https://www.youtube.com/watch?v=7RXx7zMMmEg](url)

Source file (cut off) which is causing the issue - 
[https://mega.nz/#!L09ARRLZ!xsdm_MO6TiQT4ju3mPpGIlNHkqg12ipDL3DKz-bk5mA](url)

Hardware is RPi v2, video source is located on SMB share (over LAN). I am using HDMI audio pass through since i have HW receiver Onkyo HT-S3705 which is fully capable of audio decode.

Version is "Kodi 15.2 Git:2015-12-04-858ee60 (Compiled: Dec 5 2015)". I've also did xbian update so i have latest available versions of the components.

The issue occurs only with this specific file, but when i play this on my PC via Zoom Player (with FFmpeg) everything works fine, no issue.

Let me know if more info is needed. 

Thanks in advance for taking a look on this :]

mkreisl, 2016-01-31:
Tested your sample with all my available devices - all of them plays it without any issue.
* i.mx6 - kodi 16.0 rc2
* rpi2 - kodi 16.0 rc2
* rpi1 - kodi 15.2 (same Kodi version you're using)

But I can't use audio passtrough, because my TV does not support DTS, only AC3

Does this sample also stop playing?

Stunherald, 2016-01-31:
Yop, i've tested the provided sample as well. The second log is actually from playing only the sample. I will try to disable passthrough and test again.

Stunherald, 2016-01-31:
Confirmed, cause is DTS passthrough. When i disable it the sample play without issues. When i enabled the DTS passthrough again it got stuck again.

Stunherald, 2016-02-24:
Any idea what can be done with this except disabling the DTS in Kodi? Any additional info/logfiles or whatever needed?

mkreisl, 2016-03-04:
Please try Kodi 16.0

Stunherald, 2016-03-18:
Hello,

I've just updated to Kodi 16 and unfortunately i have the same issues with DTS pass-through to my receiver.

I have another video which is "broken" from the start, no sound at all, playback acts "weird". When the DTS pass-through is disabled everything works perfectly. I can provide fresh logs + sample file again if you want.

Thank you in advance for your help!

popcornmix, 2016-03-19:
@Stunherald Can you test with this firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_dtshang.zip

extract the files to the boot partition (the one you see from windows).

Stunherald, 2016-03-21:
@popcornmix : Eeeeeh, sh!t ... ^^

[https://i.imgur.com/TzVJaJI.png](url)

popcornmix, 2016-03-21:
Just copy the ``start*.elf`` and ``fixup*.dat`` that currently exist on the sdcard and not the ones that aren't currently there. You can move the ``start*.elf`` and ``fixup*.dat`` off the sdcard first.

Stunherald, 2016-03-21:
@popcornmix : Bingo! You fixed it ^^ I tried the sample mentioned in first post and it works! Thanks a million!

closed 2017-01-27 by mkreisl