xbianonpi/xbian #808: cubox: CEC doesn't work any more after latest update

didlich, 2016-02-02:
Updated xbian yesterday on my cubox, now there is no CEC on my Samsung TV any more

stopping xbmc and using cec-client finds my tv, also the remote commands are detected, but xbmc has this in the log:
```
21:36:34  21.702358 T:1953661504   ERROR: Unable to load libcec.so.3.0, reason: libcec.so.3.0: cannot open shared object file: No such file or directory
21:36:34  21.702490 T:1953661504   ERROR: Initialise - failed to initialise bus cec                      
21:36:34  21.825460 T:1686107056  NOTICE: Thread PeripBusUSBUdev start, auto delete: false                 
21:36:34  22.101448 T:1953661504   ERROR: DBus: Error org.freedesktop.UDisks.Error.PermissionDenied - Not Authorized
21:36:34  22.159216 T:1953661504   ERROR: DBus: Error org.freedesktop.UDisks.Error.Failed - Error mounting: mount exited with exit code 1: helper failed wit
```

mkreisl, 2016-02-02:
Do you have staging repo enabled?

didlich, 2016-02-02:
I think I have also staging:
```
cat /etc/apt/sources.list.d/xbian.list
```
```
deb mirror://apt.xbian.org/mirror.txt stable main imx6-jessie
deb mirror://apt.xbian.org/mirror.txt staging main imx6-jessie
```

mkreisl, 2016-02-02:
Ok. So you have 2 choices:
a) reinstall CEC packages from stable repo
or
b) install Kodi Jarvis 16.0 rc3 from staging
Unfortunately, the package name has been changed from xbian-package-xbmc-6q to xbian-package-xbmc, to install please run `apt-get install xbian-package-xbmc`

didlich, 2016-02-03:
thanks,
I used the first option and removed the staging repos, then:
```
dpkg -r --force-all cec-utils libcec libcec-dev libcec3:armhf python-libcec
apt-get -f install
```
after a reboot the problem is solved

closed 2016-02-08 by didlich