xbianonpi/xbian #809: cubox: vdr 2.2.0~xbian-1446330623

didlich, 2016-02-03:
new vdr (old: 2.0.3-3) packages are available in xbian, but it seems they are not working
is someone using the new vdr packages on the cubox?

mkreisl, 2016-02-03:
What exactly does not work? 
I'm using the vdr 2.2.0~xbian package for RPi2 (this is my PVR backup system :smile:) and works very well there.

didlich, 2016-02-03:
after installing the packages I tried to start vdr by:
```
sudo start vdr
```
but nothing happens

then I tried to execute the binary by: 
**/usr/bin/vdr**

it it tells me: **illegal instruction**

mkreisl, 2016-02-03:
Hmmm, sounds not good. Will test it next days on my Hummingboard-Pro

didlich, 2016-02-03:
ok, it would be fine, if you could give some feedback

mkreisl, 2016-02-04:
Found the culprit :smile: New packages are online

didlich, 2016-02-04:
thanks, I'll try it.

didlich, 2016-02-08:
today I had time to test, it works

I installed also a new set of vdr-plugins, but the vdr-plugin-live is not working
@mkreisl  do you also use the vdr live plugin with vdr vesion 2.2?

closed 2016-02-08 by didlich

mkreisl, 2016-02-08:
@didlich Currently I'm not using vdr because I was switching to TVheadend some weeks ago. It has better support of the EPG EIT - but IMO timeshift works still better with vdr I used many years before.
Live plugin worked - not all functions - but was ok for me.

didlich, 2016-02-11:
@mkreisl thanks for your response
while reading the new configuration concept of vdr-2.2 I found finally the problem with vdr-plugin-live, there was a missing symbolic link in /etc/vdr/conf.d for live plugin
so the solution is:
```
cd /etc/vdr/conf.d/
ln -s ../conf.avail/live.conf
```

may be you can fix this in the next build?

mkreisl, 2016-02-12:
@didlich
Thank you for your response :+1: 
A missing `!` in postinst script was the culprit for some addons not to be autoenabled after installation. Will fix this in the next days