xbianonpi/xbian #810: Feature request: rtk8192eu

djcf, 2016-02-09:
Hi there,

I built a blob for the realtek 8192eu wireless device for latest Xbian 4.1.13+. Is there any interest in including the driver in the distribution?

https://github.com/djcf/rtl8192eu-xbian

best,

Daniel

mk01, 2016-02-12:
@djcf 

sure, ... please transform the howto into hook script like this - this way it will be always compiled (and packed) into any kernel package be build.
thanks!

https://github.com/xbianonpi/xbian-package-kernel/blob/master/hooks.d/pre-deb.d/rtl8812au

mkreisl, 2016-02-12:
@mk01 
Already did this, only have to push my changes :smile:

mkreisl, 2016-02-12:
@djcf 
New kernel packages are created with included rtl8192eu driver. Could you please test it?
Thanks

closed 2016-03-23 by mkreisl