xbianonpi/xbian #812: Kodi 16 not starting - assertion failure

rupaschomaker, 2016-03-03:
Rpi 2, running staging though saw the same with stable:
 
```
xbian@xbiankids /var/log/upstart $ sudo cat xbmc.log
kodi.bin: /home/xbian/xbian-package-xbmc/build/rpi2-jarvis/working/xbmc/utils/MathUtils.h:65: int MathUtils::round_int(double): Assertion `x > static_cast<double>((int) ((-2147483647 - 1) / 2)) - 1.0' failed.
Aborted
start: Job failed to start
stop: Job has already been stopped: xbmc
```
```
xbian@xbiankids ~/.kodi/temp $ cat kodi.log
00:04:45 230.219574 T:1954979840  NOTICE: special://profile/ is mapped to: special://masterprofile/
00:04:45 230.219910 T:1954979840  NOTICE: -----------------------------------------------------------------------
00:04:45 230.220139 T:1954979840  NOTICE: Starting Kodi (16.0 Git:2016-03-02-889e631). Platform: Linux ARM (Thumb) 32-bit
00:04:45 230.220291 T:1954979840  NOTICE: Using Debug Kodi x32 build (version for Raspberry Pi)
00:04:45 230.220428 T:1954979840  NOTICE: Kodi compiled Mar  3 2016 by GCC 4.9.2 for Linux ARM (Thumb) 32-bit version 4.1.7 (262407)
00:04:45 230.220978 T:1954979840  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 4.4.0+
00:04:45 230.221146 T:1954979840  NOTICE: FFmpeg statically linked, version: 2.8.6-kodi-2.8.6-Jarvis-16.0
00:04:45 230.221283 T:1954979840  NOTICE: Host CPU: ARMv7 Processor rev 5 (v7l), 4 cores available
00:04:45 230.221405 T:1954979840  NOTICE: ARM Features: Neon enabled
00:04:45 230.221527 T:1954979840  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
00:04:45 230.221664 T:1954979840  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
00:04:45 230.221786 T:1954979840  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
00:04:45 230.221924 T:1954979840  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
00:04:45 230.222107 T:1954979840  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
00:04:45 230.222321 T:1954979840  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
00:04:45 230.222519 T:1954979840  NOTICE: Local hostname: xbiankids
00:04:45 230.222656 T:1954979840  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
00:04:45 230.222839 T:1954979840  NOTICE: -----------------------------------------------------------------------
00:04:46 230.253036 T:1954979840    INFO: Selected UPower as PowerSyscall
00:04:46 230.316681 T:1954979840   DEBUG: DeviceKit.Power: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.DeviceKit.Disks was not provided by any .service files
00:04:46 230.317062 T:1954979840    INFO: Selected Logind/UPower as PowerSyscall
00:04:46 230.339096 T:1954979840   DEBUG: LogindUPowerSyscall - inhibit lock taken, fd 14
00:04:46 230.350449 T:1954979840    INFO: Selected UPower as PowerSyscall
00:04:46 230.360428 T:1954979840   DEBUG: LogindUPowerSyscall - delay lock released
00:04:46 230.361053 T:1954979840  NOTICE: load settings...
00:04:46 230.452637 T:1954979840   DEBUG: CSettings: loaded settings definition from special://xbmc/system/settings/settings.xml
00:04:46 230.500656 T:1954979840   DEBUG: CSettings: loaded settings definition from special://xbmc/system/settings/rbp.xml
00:04:46 230.503860 T:1954979840   DEBUG: CSettings: loaded settings definition from special://xbmc/system/settings/rbp2.xml
00:04:46 230.506729 T:1945105328   DEBUG: Thread FDEventMonitor start, auto delete: false
00:04:46 230.535049 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround71" for playback
00:04:46 230.537888 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround51" for playback
00:04:46 230.538239 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround71" for playback
00:04:46 230.540634 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround40" for playback
00:04:46 230.540894 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround51" for playback
00:04:46 230.541107 T:1954979840    INFO: CAESinkALSA - Unable to open device "surround71" for playback
00:04:46 230.594681 T:1954979840  NOTICE: Found 2 Lists of Devices
00:04:46 230.594925 T:1954979840  NOTICE: Enumerated PI devices:
00:04:46 230.595062 T:1954979840  NOTICE:     Device 1
00:04:46 230.595688 T:1954979840  NOTICE:         m_deviceName      : HDMI
00:04:46 230.595856 T:1954979840  NOTICE:         m_displayName     : HDMI
00:04:46 230.595978 T:1954979840  NOTICE:         m_displayNameExtra:
00:04:46 230.596207 T:1954979840  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
00:04:46 230.596329 T:1954979840  NOTICE:         m_channels        : FL,FR
00:04:46 230.596527 T:1954979840  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
00:04:46 230.596664 T:1954979840  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
00:04:46 230.596848 T:1954979840  NOTICE:     Device 2
00:04:46 230.597061 T:1954979840  NOTICE:         m_deviceName      : Analogue
00:04:46 230.597198 T:1954979840  NOTICE:         m_displayName     : Analogue
00:04:46 230.597321 T:1954979840  NOTICE:         m_displayNameExtra:
00:04:46 230.597458 T:1954979840  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
00:04:46 230.597580 T:1954979840  NOTICE:         m_channels        : FL,FR
00:04:46 230.597717 T:1954979840  NOTICE:         m_sampleRates     : 48000
00:04:46 230.597839 T:1954979840  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
00:04:46 230.597961 T:1954979840  NOTICE:     Device 3
00:04:46 230.598221 T:1954979840  NOTICE:         m_deviceName      : Both
00:04:46 230.598358 T:1954979840  NOTICE:         m_displayName     : HDMI and Analogue
00:04:46 230.598480 T:1954979840  NOTICE:         m_displayNameExtra:
00:04:46 230.598602 T:1954979840  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
00:04:46 230.598724 T:1954979840  NOTICE:         m_channels        : FL,FR
[... 99 lines cut ...]
                                            ramfsaddr=0x1000000
                                            sdram_freq=500
                                            temp_limit=85
00:04:47 231.443558 T:1954979840  NOTICE: Config:
                                            device_tree=-
                                            ramfsfile=initramfs.gz
00:04:47 232.027252 T:1954979840  NOTICE: Register - new video device registered on platform->/sys/class/graphics/fb0: generic_video,hdmi_video (0000:0000)
00:04:47 232.027588 T:1954979840  NOTICE: InitWindowSystem: Using EGL Implementation: raspberrypi
00:04:47 232.043365 T:1954979840 WARNING: CalcSAR: Screen SAR - not usable info
00:04:47 232.043808 T:1954979840  NOTICE: Found resolution 720x480 (720x480) @ 60.00i - Full Screen
00:04:47 232.044174 T:1954979840  NOTICE: Checking resolution 16
00:04:47 232.059860 T:1954979840  NOTICE: GL_VENDOR = Broadcom
00:04:47 232.060120 T:1954979840  NOTICE: GL_RENDERER = VideoCore IV HW
00:04:47 232.060257 T:1954979840  NOTICE: GL_VERSION = OpenGL ES 2.0
00:04:47 232.060394 T:1954979840  NOTICE: GL_SHADING_LANGUAGE_VERSION = OpenGL ES GLSL ES 1.00
00:04:47 232.060547 T:1954979840  NOTICE: GL_EXTENSIONS = GL_OES_compressed_ETC1_RGB8_texture GL_OES_compressed_paletted_texture GL_OES_texture_npot GL_OES_depth24 GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_EGL_image_external GL_EXT_discard_framebuffer GL_OES_rgb8_rgba8 GL_OES_depth32 GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_APPLE_rgb_422 GL_EXT_debug_marker
00:04:48 232.764374 T:1954979840  NOTICE: Running database version Addons20
00:04:48 232.768082 T:1954979840  NOTICE: Running database version ViewModes6
00:04:48 232.772079 T:1954979840  NOTICE: Running database version Textures13
00:04:48 233.145645 T:1954979840  NOTICE: Running database version xbmc_music56
00:04:49 233.256470 T:1954979840  NOTICE: Running database version xbmc_video99
00:04:49 233.261017 T:1954979840  NOTICE: Running database version TV29
00:04:49 233.265137 T:1954979840  NOTICE: Running database version Epg11
00:04:49 233.268997 T:1954979840  NOTICE: start dvd mediatype detection
```

mkreisl, 2016-03-03:
Do you overclocking your RPi2?

rupaschomaker, 2016-03-03:
I was, but I turned off overclocking and tried again.  Same problem.

rupaschomaker, 2016-03-05:
I upgraded another RPI2 and it doesn't have this problem.  Gonna close this one out.

closed 2016-03-05 by rupaschomaker

vondrt4, 2016-04-01:
I'm having this issue on my Raspberry Pi 1. Any clue about what may be causing it?

rupaschomaker, 2016-04-06:
@vondrt - I have two rpi2 devices. One shows this and one doesn't. Only wanted kodi on one (use Nexus player for watching, RPI for indexing and dB maintenance) so use the good one.

mkreisl, 2016-04-06:
Never seen this. No idea how I could help you.

Perhaps you could rename folder **/home/xbian/.kodi** so kodi starts with an empty configuration

vondrt4, 2016-04-06:
I only have one piece and there is a MPEG2 license on it, so I would really like to keep it as long as there is MPEG2 DVB-T in this country. I could try re-imaging, but I'd be very angry if that helped, because I have some mods on it and would have to collect them all :-|.

What I've read, it may be absolutely anything, but the reports are mainly concerned with wrong or missing dependencies. What has changed in that regard since 15.2, which works for me?
And could the new version have greater requirements e.g. on video RAM? I have set it to 64 MB as I only do SDTV. 50 MB of ZRAM .kodi/temp will do?

mkreisl, 2016-04-06:
> I could try re-imaging it, but I'd be very angry if that helped, because I have some mods on it, most importantly an unsupported build of MythTV, and would have to collect them all and install them again :-|.

Hopefully you'll have a proper backup :smile: 

Re-imaging is IMO not necessary. Try moving your .kodi and see what happens. And increase your video RAM (see below)

> And could the new version have greater requirements e.g. on video RAM? I have set it to 64 MB as I only do SDTV.

That' a very low value. I'm using 128MB on my RPi1's

> 50 MB of ZRAM on .kodi/temp will do?
32MB here

rupaschomaker, 2016-04-06:
Are you seeing it fail for the exact same reason as I opened on this issue?

vondrt4, 2016-04-09:
Yes, I believe so. It's exactly the same assert failure and the process kills itself with SIGABRT. I tried adding video ram and disabling overclock, but it didn't help. I've run it as root -> without configuration, and it did the same. The log has nothing interesting in it.

mkreisl, 2016-04-09:
@vondrt4 
would be very helpful if you make an image from your installation (**xbian-config->xbian copier**) for investigation

mkreisl, 2016-04-10:
@vondrt4 
flashed your image and after some testing found the culprit for your issue:
You are using composite output / sdtv

It seems that sdtv is not supported very well since kodi 16. I got it running on my Rpi1 installation, but videos are always in wrong aspect ratio (getting 'egg heads'). When using 'vanilla' kodi 16, AR is also wrong, but getting no 'egg heads', getting 'fat heads' now.
Already tested kod 17, getting more issues: wrong aspect ratio and omx player always displaying black window.

So it seems you'll have to stay on 15.2 and waiting for any solution (but can't promise that there will be an solution in future)

vondrt4, 2016-04-10:
That's not nice. So no more bleeding edge for me :-|. Well, there won't be much development for the Pi 1, will it?
While you have the image, could you have a look at a few other bad things my instalation developed?
a) Kodi 15.2 crashes when you open a long running dialog, e.g. language selection or theme selection. It may have something to do with the fact I enabled the hardware watchdog. That was because of the Afatech 9015 DVB-T dongle, which has some problems with USB transfers dropping out, after which the whole USB subsystem sometimes freezes, about once a week. You helped me debug that by building a kernel with Netconsole, if you remember.
It also reboots when I select Exit Kodi in its main screen.
b) xbian-config, the curses set-up program, is broken. There are menu entries interleaved with numbers. And I did upgrade regularly.
c) And have a look at the way I modified the xbmc start-up scripts. They are disabled and there is a curses prompt called askxbmc.sh spawned on the first console which lets you (by default, just by pressing Enter) launch Kodi, or drop to the shell (and use xinit from there). I think it is a good idea. My Raspberry mostly runs as MythTV backend and more CPU usage from Kodi means more dropped USB frames, so I launch it only when I need it.

All in all, I have found that the Raspberry Pi is a bad choice for a TV headend / PVR. Is the version 2 or 3 better? If you, e.g., dedicate one core to interrupts, it could work without dropouts..

vondrt4, 2016-05-07:
@mkreisl 
> It seems that sdtv is not supported very well since kodi 16.
You were talking about wrong aspect ratios, but my problem is that I cannot launch Kodi 16 at all.

Progress: Flashed a clean Xbian image - the same error.
Flashed a clean image of two major competitors od Xbian - both worked.
Finally: Installed Kodi 16.1 from OSMC over Xbian - the same firmware, kernel, library environment - works. You build of Kodi must have something set the wrong way.

And I have tried playing a 16:9 and 4:3 movie on my 4:3 TV and had no aspect ratio problems.

mkreisl, 2016-05-09:
@vondrt4 
Please try this version http://apt.xbian.org/pool/devel/rpi-jessie/x/xbian-package-xbmc/xbian-package-xbmc_16.1-1462796680_armhf.deb and report back.

This version works in standard environment (1080p HDMI) and should work with SDTV also (can not test it with RPi1, but build for RP2 works here)

mkreisl, 2016-05-10:
This is test version for RPi2 and RPi3
http://apt.xbian.org/pool/devel/rpi2-jessie/x/xbian-package-xbmc/xbian-package-xbmc_16.1-1462731607_armhf.deb

vondrt4, 2016-05-12:
Yes, this version starts fine, plays video with correct aspect ratio (on 4:3 TV), and crashed once on some Czech TV video plugin that blocked for too long (with watchdog). Overall, it's not worse than 15.2 :-).