xbianonpi/xbian #813: rpi3 support

bairdy, 2016-03-03:
are we able to support the device on the rpi2 image like other distros until the 64bit support is properly introduced?

ive tried updating the firmware (rebuilt the firmware package) but im just getting a black screen on boot and works fine on rpi2 but on rpi3 I just get the power light solid (not getting the flashing green light to say the firmware is missing like i was just moving the sd card from rpi2 to rpi3) Ive also added the rpi3 dtb to boot partition but it doesnt help 

Is a kernel tweak needed to add support for the cpu in the kernel or something different? I would take a look but know next to nothing about the kernel or the building it or if anyone knows what teaks are  need i will do them and give them a try.

mkreisl, 2016-03-03:
Yes, it's one of my next task. A new kernel is needed which I have already built.

Can you please download the kernel package from here http://apt.xbian.org/pool/devel/rpi2-jessie/l/linux-image-bcm2836/,  install the package and report what happens?

Thanks

(Please note that Ethernet driver is built as module)

The-Exnor, 2016-03-03:
Well as soon i have a unit i will be glad to help you guys :)

bairdy, 2016-03-03:
Just installed and booted in KODI fine on rpi2 but the network wast working when inserted on rpi3 delving into it, it had called the interface eth1 not eth0 and also now have a one called sit0@NONE 

also will add it booted probably twice as fast as pi2 :)

mkreisl, 2016-03-03:
Sounds good :smile: 
could you please post the output of /proc/cpuinfo?

bairdy, 2016-03-03:
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a02082
Serial		: 000000005f5dc69b
```

mkreisl, 2016-03-03:
Ok, still BCM2709. Wow, BogoMIPS  has been doubled. 

So I can build a ready to use image tonight :smile:

mkreisl, 2016-03-03:
> delving into it, it had called the interface eth1 not eth0 

Because you RPI3 has a new MAC address, so a new ethX device will be created

> and also now have a one called sit0@NONE

The onboard WLAN?

bairdy, 2016-03-03:
also the firmware package i built and using is at  https://www.dropbox.com/s/h6t56hu8o7w4idp/xbian-package-firmware3.0-1457018057.deb?dl=0 to save building if need be. 

> Ok, still BCM2709. Wow, BogoMIPS has been doubled.

you can tell i'm using my old tried and tested rpi2 image and boots a lot faster and its more resposive in KODI itslef

> The onboard WLAN?

just going to find out now, is the BCM43438 wireless built into the kernel?

bairdy, 2016-03-03:
ahh sit0 is the tunneling protocol for using IPv6 over an existing IPv4 connection not the wireless

mkreisl, 2016-03-03:
> just going to find out now, is the BCM43438 wireless built into the kernel?

All wlan drivers are built as modules

bairdy, 2016-03-03:
can confirm everything seems to work apart from the wifi/bluetooth  the modules load but iwconfig tells me no wireless modules found 

lsmod output
```
Module                  Size  Used by
zram                   20264  1 
zsmalloc                9901  1 zram
bnep                    9936  2 
bluetooth             316664  5 bnep
joydev                  8885  0 
evdev                  10787  3 
brcmfmac              175111  0 
brcmutil                5585  1 brcmfmac
hid_microsoft           3207  0 
smsc95xx               16648  0 
cfg80211              407422  1 brcmfmac
rfkill                 14525  3 cfg80211,bluetooth
bcm2835_gpiomem         2939  0 
bcm2835_wdt             3128  0 
lirc_rpi                6170  3 
lirc_dev                7922  1 lirc_rpi
rc_core                16040  1 lirc_dev
uio_pdrv_genirq         3025  0 
uio                     7907  1 uio_pdrv_genirq
uinput                  7105  1 
snd_bcm2835            23699  0 
snd_pcm                71756  1 snd_bcm2835
snd_timer              18385  1 snd_pcm
snd                    49527  3 snd_bcm2835,snd_timer,snd_pcm
```

bairdy, 2016-03-03:
and dmesg log

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.3+ (manfred@kmnote5.kmhome.com) (gcc version 4.9.3 20141031 (prerelease) (Linaro GCC 2014.11) ) #1 SMP Tue Mar 1 20:32:22 CET 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x2e400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 192512
[    0.000000] free_area_init_node: node 0, pgdat 809fdfc0, node_mem_map add64000
[    0.000000]   Normal zone: 1692 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 192512 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9540->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @aefaa000 s22528 r8192 d22528 u53248
[    0.000000] pcpu-alloc: s22528 r8192 d22528 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 190820
[    0.000000] Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2709.boardrev=0xa02082 bcm2709.serial=0x5f5dc69b smsc95xx.macaddr=B8:27:EB:5D:C6:9B bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  telnet zswap.enabled=1 zswap.compressor=lz4 console=tty1 root=/dev/mmcblk0p2 rootflags=subvol=root/@,autodefrag,compress=lz4 rootfstype=btrfs rootwait logo.nologo quiet noswap loglevel=0 partswap startevent=mountall selinux=0 splash nohdparm --startup-event mountall
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 742912K/770048K available (7387K kernel code, 463K rwdata, 1872K rodata, 512K init, 791K bss, 18944K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xaf800000 - 0xff800000   (1280 MB)
    lowmem  : 0x80000000 - 0xaf000000   ( 752 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80913168   (9261 kB)
      .init : 0x80914000 - 0x80994000   ( 512 kB)
      .data : 0x80994000 - 0x80a07f28   ( 464 kB)
       .bss : 0x80a0a000 - 0x80acfc5c   ( 792 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000005] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000014] Switching to timer-based delay loop, resolution 52ns
[    0.000176] Console: colour dummy device 80x30
[    0.000283] console [tty1] enabled
[    0.000304] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000316] pid_max: default: 32768 minimum: 301
[    0.000496] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000503] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001054] Disabling cpuset control group subsystem
[    0.001071] Initializing cgroup subsys io
[    0.001088] Initializing cgroup subsys memory
[    0.001110] Initializing cgroup subsys devices
[    0.001121] Initializing cgroup subsys freezer
[    0.001132] Initializing cgroup subsys net_cls
[    0.001158] CPU: Testing write buffer coherency: ok
[    0.001198] ftrace: allocating 23195 entries in 69 pages
[    0.030110] CPU0: update cpu_capacity 1024
[    0.030131] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.030139] [bcm2709_smp_prepare_cpus] enter
[    0.030226] Setting up static identity map for 0x8240 - 0x8274
[    0.031181] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.031446] [bcm2709_secondary_init] enter cpu:1
[    0.031472] CPU1: update cpu_capacity 1024
[    0.031475] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.031678] [bcm2709_boot_secondary] cpu:2 started (0) 18
[... 238 lines cut ...]
[   26.146092] zram0: detected capacity change from 0 to 134217728
[   26.189352] Adding 131068k swap on /dev/zram0.  Priority:20 extents:1 across:131068k SSFS
[   26.298329] Adding 250876k swap on /dev/mmcblk0p3.  Priority:0 extents:1 across:250876k SSFS
[   27.321442] systemd-logind[1990]: Failed to start user service: Unknown unit: user@0.service
[   27.338309] systemd-logind[1990]: New session c1 of user root.
[   49.517735] systemd-logind[1990]: New session c2 of user root.
```

which look nasty around here 

```
[   10.339606] brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
[   10.349667] usbcore: registered new interface driver brcmfmac
[... cut ...]
[   10.909928] lirc_rpi lirc_rpi: lirc_dev: driver lirc_rpi registered at minor = 0
[   10.909943] lirc_rpi: driver registered!
[   11.252364] smsc95xx 1-1.1:1.0 eth1: hardware isn't capable of remote wakeup
[   11.252686] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   11.359272] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   12.130475] input: lircd as /devices/virtual/input/input2
[   12.377147] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

bairdy, 2016-03-03:
seems that the firmware file is missing i got a copy of the firmware file from https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm80211 and copied the brcm folder and contents to /lib/firmware and  rebooted and seems to work fine but i have to bring the interface up manually

bairdy, 2016-03-03:
managed to fix my issues removing 
```
/etc/udev/rules.d/70-persistent-net.rules
```
fixed the udev naming the network adapters wrong and reset the names

and as for the wireless not starting was down to my
```
/etc/network/interfaces
```
it didn't like having the wireless adapter and ethernet both on static ip's (both configured by xbian-config) so just a config issue there. will have a look at the bluetooth tomorrow and see it that works once i buy a keyboard

mkreisl, 2016-03-04:
You did a great work an helped me a lot :+1: 
Preliminary image is ready for download http://www.xbian.org/preliminary-raspberry-pi-3-support/

bairdy, 2016-03-04:
no worries will give a quick try just a shame sourceforge if a bit crappy on dl speeds

bairdy, 2016-03-04:
the image boots fine except starting in 640x480 (but could be my denon reciever sending duff res info), but just a slight issue with the wireless firmware the brcmfmac43430-sdio.txt needs to included in the firmware folder aswell as the brcmfmac43430-sdio.bin to work but once placed into the folder all works fine

mkreisl, 2016-03-04:
Great. Next image will be available tomorrow :smile:

> the image boots fine except starting in 640x480 (but could be my denon reciever sending duff res info),

That's normal. Is the standard behavior for fresh installations

bairdy, 2016-03-04:
the latest kernel (linux-image-bcm2836_4.4.3+-1457058983)  in devel has fixed the ip6 to ipv tunnel module being loaded and not being able to be blacklisted

bairdy, 2016-03-09:
I have tried getting the bluetooth to work over the last couple of days but have ran into a couple  issues.
 
after installing bluez and bluetooth

firstly the firmware (BCM43438A1.hcd) is missing i made a copy from a foundation image to try and get to work but then it still doesnt load. after digging around the foundation have made a pi-blutooth package which contains a systemd startup script with
```
[Unit]
Description=Configure Bluetooth Modems connected by UART
ConditionPathIsDirectory=/proc/device-tree/soc/gpio@7e200000/bt_pins
Before=bluetooth.service
After=dev-ttyAMA0.device

[Service]
Type=forking
ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

[Install]
WantedBy=multi-user.target
```
to start it, but when i have tried to start it i get the following
```
Failed to get D-Bus connection: Unknown error -1
```
so tried starting manually using the command in the script and get the following
```
bcm43xx_init
Cannot open directory '/etc/firmware': No such file or directory
Patch not found, continue anyway
Set Controller UART speed to 921600 bit/s
```
which after making dir a link to /lib/firmware and having a reboot systemd still wont execute the service, but after manually doing 
```
/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -
```
i got 
```
bcm43xx_init
Flash firmware /etc/firmware/BCM43430A1.hcd
Initialization timed out.
```
so gave it another try and got 
```
bcm43xx_init
Patch not found, continue anyway
Set Controller UART speed to 921600 bit/s
Device setup complete
```
and then ran 
```
hcitool dev
```
and it now has given us
```
Devices:
        hci0    B8:27:EB:F7:6C:31
```
so i now have got it working maually :smiley: 
and also double checking using bluetoothctl gives
```
[bluetooth]# show
Controller B8:27:EB:F7:6C:31
        Name: xbian
        Alias: xbian
        Class: 0x000000
        Powered: yes
        Discoverable: no
        Pairable: yes
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0517
        Discovering: no
```
and after a scan it manages to find my phone
```
root@xbian ~ # hcitool scan
Scanning ...
        C0:EE:FB:35:17:CB       OnePlus One
```
so the issues faced are 

1. missing firmware

2 a start up script that works

3 hciattch looking for firmware in the wrong dir (rpi foundation must be using a patch as theres no system link to here)

mkreisl, 2016-03-09:
@bairdy 
Thank for reporting this, it's really an excellent explanation what has to be done. :+1: 
With all these information, it should be possible to make it work.

But, it's confusing me that you have to run `hciattach` twice

Btw: XBian uses Upstart instead of Systemd, so all systemd startup scripts will never work

mkreisl, 2016-03-09:
@bairdy 
Just hacked an upstart start script (took it from i.MX6 installation). Can you please try it?

http://paste.debian.net/413667/

```
start on stopped udevtrigger and stopped module-init-tools

script
    [ $(xbian-arch) = RPI ] || exit 0

    [ -e /dev/ttyAMA0 ] || { echo "/dev/ttyAMA0 not there, waiting 5s"; sleep 5; }
    [ -e /proc/device-tree/soc/gpio@7e200000/bt_pins ] || echo "/proc/device-tree/soc/gpio@7e200000/bt_pins not there, isn't this RPi3?"

    [ -e /etc/firmware ] || ln -s /lib/firmware /etc/firmware
    
    /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - || :
    sleep 3
    /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - || :

end script

post-start script
    [ $(xbian-arch) = RPI ] || exit 0

    if [ -z "$(ls /sys/class/bluetooth 2>/dev/null)" ]; then
        [ ! -e /etc/default/bluetooth ] || sed -i 's/BLUETOOTH_ENABLED=./BLUETOOTH_ENABLED=0/' /etc/default/bluetooth
        for d in $(ls /sys/class/bluetooth 2>/dev/null); do
            hciconfig $d up || :
        done
    else
        [ ! -e /etc/default/bluetooth ] || sed -i 's/BLUETOOTH_ENABLED=./BLUETOOTH_ENABLED=1/' /etc/default/bluetooth
    fi

end script

post-stop script
    [ $(xbian-arch) = RPI ] || exit 0

    if [ -z "$(ls /sys/class/bluetooth 2>/dev/null)" ]; then
        [ ! -e /etc/default/bluetooth ] || sed -i 's/BLUETOOTH_ENABLED=./BLUETOOTH_ENABLED=0/' /etc/default/bluetooth
    else
        [ ! -e /etc/default/bluetooth ] || sed -i 's/BLUETOOTH_ENABLED=./BLUETOOTH_ENABLED=1/' /etc/default/bluetooth
    fi
end script
```
Store it to fle **/etc/init/rpi3-bluetooth** and then run `start rpi3-bluetooth`
Watch/paste the output of the script in the logfile `/var/log/upstart/rpi3-bluetooth.log`

Thanks

anaconda, 2016-03-09:
There are a few patches from RPF to increase firmware load timeout and fix the firmware directory:
https://gist.github.com/pelwell/c8230c48ea24698527cd

mkreisl, 2016-03-09:
@anaconda 
Yeah, those patches makes sense.

Seems that source package of bluez differs from binary package

Thanks for the link

mkreisl, 2016-03-09:
bluez package patched and uploaded

bairdy, 2016-03-09:
thanks @mkreisl just aved me a job, was just about to patch will install and give a try.

as for trying hcattach twice im not to sure i was typing as i was doing trial and error so i knew where i was at then just copied and pasted here, i normally double check over what i have typed

mkreisl, 2016-03-09:
@bairdy
this patched bluez package should solve this issue

bairdy, 2016-03-09:
as for the version of bluez it's working perfect just the script wont work but starts manually fine and doesn't need the firmware folder in /etc and loads with no issues

mkreisl, 2016-03-09:
could you please post the log /var/log/upstart/rpi3-bluetooth.log

bairdy, 2016-03-09:
rpi3-bluetooth log
```
start: Unknown job: on
start: Unknown job: on
start: Unknown job: on
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
bcm43xx_init
Failed to write reset command
bcm43xx_init
Failed to write reset command
Can't initialize device: Success
bcm43xx_init
Failed to write reset command
Can't initialize device: Success
```

mkreisl, 2016-03-09:
```
start: Unknown job: on
start: Unknown job: on
start: Unknown job: on
```
I'm confused. Where does this came from ???

When I start this script on my RPi2, I don't get this strange message

bairdy, 2016-03-09:
not sure but after a restart ive got 

```
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
bcm43xx_init
Failed to write reset command
Can't initialize device: Success
```
but with the script i'm also not able to find any devices

mkreisl, 2016-03-09:
maybe it does not like the double start of hciattach

bairdy, 2016-03-09:
posibbly not removing the extra start looks better in the log
```
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
```
and allows discovery of devices again

```
root@xbian ~ # hcitool scan
Scanning ...
        C0:EE:FB:35:17:CB       OnePlus One
        C4:57:6E:81:AA:E4       [TV]Samsung LED55
```

idarek, 2016-03-12:
Is it normal, that RPi3 report itself in dmesg as bcm2709 and not bcm2710 ?

bairdy, 2016-03-12:
yeah it is usual even on the foundations raspbian images it reports as that

singhprd, 2016-03-13:
+1 for bluetooth support. Thanks all for the work everyone!

mkreisl, 2016-03-13:
@singhprd 
Are you using this start script ( http://paste.debian.net/413743 ) I posted here?

closed 2017-01-27 by mkreisl