xbianonpi/xbian #816: Wrong time and date.

BluOrc, 2016-03-29:
Every time I turn off or restart my raspberry the date and time become wrong. 
I already configured in:

settings > appearance > International > region (Brazil / America / Bahia)
settings > xbian > system > timezone (Brazil / America / Bahia)

The only way to get the correct date and time and forcing a updating with "sudo ntpd -gq" but simply restart or shut down they come back to get wrong.

OS: Xbian 1.0 (knockout) (kernel: Linux 4.1.13+)
Ver: Kodi 16.0 Git:2016-03-04f47fb5e

coderpop, 2016-03-29:
As a quick fix you could put sudo "ntpd -gq" into /etc/rc.local so it runs automatically every time the pi starts up.
Is ntpd running OK?

BluOrc, 2016-03-29:
I did a fresh install and configure tried by Kodi and "sudo xbian-config" both are wrong.

[image]

"ntp" is running but "ntpd" i think no becouse dont appear in the list of services.

I try the quick fix later.

mkreisl, 2016-03-29:
Nice date (seems you are time travelling) :smile: 

I agree, ntp(d) is not able to fetch the time from timeserver (ntp is the correct service name). Maybe try different time servers (see file /etc/ntp.conf)

pille5, 2016-12-28:
I have the same problem. I updated the `/etc/ntp.conf` I've added the following servers:
```  
server 0.hu.pool.ntp.org
server 1.hu.pool.ntp.org
server 2.hu.pool.ntp.org
server 3.hu.pool.ntp.org
```

```bash
xbian@xbian ~ $ ntpq -pn
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
 193.225.50.68   .INIT.          16 u    - 1024    0    0.000    0.000   0.000
 193.225.118.163 .INIT.          16 u    - 1024    0    0.000    0.000   0.000
 163.172.138.31  .INIT.          16 u    - 1024    0    0.000    0.000   0.000
 213.136.0.252   .INIT.          16 u    - 1024    0    0.000    0.000   0.000
 193.6.176.59    .INIT.          16 u    - 1024    0    0.000    0.000   0.000
 193.224.65.147  .INIT.          16 u    - 1024    0    0.000    0.000   0.000
```

Nothing is changed after running `sudo ntpd -gq`. Any idea? How can I figure out where is the problem?

mkreisl, 2016-12-28:
@pille5 
Maybe you have to stop service (`sudo stop ntp`) before you run command

rlkollman, 2017-01-02:
I ended up editing /var/lib/ntp/ntp.conf.dhcp to reflect the correct NTP server (instead of /etc/ntp.conf).  

Not that it matters, but I guess, here's how I came to this conclusion.  Inside the /etc/ntp.conf file, the 2nd line was "driftfile /var/lib/ntp/ntp.drift" ..  Attempted to edit that file, however, it did not exist.  So, just went into that folder and the only file in there was ntp.conf.dhcp.  About the 6th line down showed the IP address NTP was using as the server, so I just changed it to my router IP and I don't have the problem anymore.

Hope this helps someone!