xbianonpi/xbian #817: Networkserver vs DNS/Nameserver in XBian-config+typo

Szurgent, 2016-03-29:
Following little bug available in the xbian-config/System/Network/selected interfaces and type=static:

-Status
-SSID
-Type
 -IP Address
 -Network
 -Gateway
 -Networkserver
-Networkserver

1: please add a <space> to the beginning of the last item.
2: Primary DNS and Secondary DNS better than Networkserver and Networkserver

I use raspberry pi 3 with updated preliminari image.

mkreisl, 2016-03-29:
Hmmm,  I have **Nameserver** there.
But you're right, a small typo is there

mkreisl, 2016-03-30:
@Szurgent 
Fixed the column alignment issue. Also the the string "Networkserver" has been changed to "Nameserver", but XBian config does not distinguish between primary and secondary name server.

Btw - you only got the string "Networkserver" if the language which has been selected is not supported

Szurgent, 2016-03-30:
I see that good in the last release.Thank you.

closed 2016-04-08 by mkreisl