xbianonpi/xbian #818: Upgrade package udev 215-17+deb8u4 failed

er0l, 2016-04-05:
Anyone facing this issue when upgrading to latest packages:

`Setting up udev (215-17+deb8u4) ...
addgroup: The group `input' already exists as a system group. Exiting.
udev stop/waiting
udev start/running, process 16885
update-initramfs: deferring update (trigger activated)
initctl: Unknown job: K01nfs-common
insserv: warning: script 'K01nfs-common' missing LSB tags and overrides
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S01dbus
insserv: warning: script 'S01dbus' missing LSB tags and overrides
initctl: Unknown job: S01dbus
initctl: Unknown job: S01dbus
initctl: Unknown job: S01dbus
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S03nfs-common
initctl: Unknown job: S03checkroot-bootclean.sh
insserv: warning: script 'S03checkroot-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S03mountnfs-bootclean.sh
insserv: warning: script 'S03mountnfs-bootclean.sh' missing LSB tags and overrides
insserv: warning: script 'udev' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `udev'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `udev'
insserv: warning: script 'noip2' missing LSB tags and overrides
initctl: Unknown job: nfs-common
insserv: warning: script 'nfs-common' missing LSB tags and overrides
initctl: Unknown job: bootmisc.sh
insserv: warning: script 'bootmisc.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `bootmisc.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `bootmisc.sh'
initctl: Unknown job: checkroot-bootclean.sh
insserv: warning: script 'checkroot-bootclean.sh' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `checkroot-bootclean.sh'
initctl: Unknown job: mountall-bootclean.sh
insserv: warning: script 'mountall-bootclean.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountall-bootclean.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountall-bootclean.sh'
initctl: Unknown job: mountall.sh
insserv: warning: script 'mountall.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountall.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountall.sh'
initctl: Unknown job: mountnfs-bootclean.sh
insserv: warning: script 'mountnfs-bootclean.sh' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountnfs-bootclean.sh'
initctl: Unknown job: mountnfs.sh
insserv: warning: script 'mountnfs.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountnfs.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountnfs.sh'
insserv: warning: script 'dbus' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `dbus'
initctl: Unknown job: mtab.sh
insserv: warning: script 'mtab.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mtab.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mtab.sh'
insserv: warning: script 'hostname.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `hostname.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `hostname.sh'
initctl: Unknown job: mountkernfs.sh
insserv: warning: script 'mountkernfs.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountkernfs.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountkernfs.sh'
initctl: Unknown job: mountdevsubfs.sh
insserv: warning: script 'mountdevsubfs.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountdevsubfs.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountdevsubfs.sh'
initctl: Unknown job: checkroot.sh
insserv: warning: script 'checkroot.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `checkroot.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `checkroot.sh'
insserv: warning: script 'avahi-daemon' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `avahi-daemon'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `avahi-daemon'
initctl: Unknown job: rmnologin
insserv: warning: script 'rmnologin' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `rmnologin'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `rmnologin'
insserv: warning: script 'cron' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `cron'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `cron'
initctl: Unknown job: ssh
insserv: warning: script 'ssh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `ssh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `ssh'
initctl: Unknown job: kmod
insserv: warning: script 'kmod' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `kmod'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `kmod'
insserv: warning: script 'lirc' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `lirc'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `lirc'
initctl: Unknown job: motd
insserv: warning: script 'motd' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `motd'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `motd'
insserv: warning: script 'openbsd-inetd' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `openbsd-inetd'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `openbsd-inetd'
insserv: warning: script 'procps' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `procps'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `procps'
initctl: Unknown job: fake-hwclock
insserv: warning: script 'fake-hwclock' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `fake-hwclock'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `fake-hwclock'
initctl: Unknown job: udev-mtab
insserv: warning: script 'udev-mtab' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `udev-mtab'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `udev-mtab'
insserv: warning: script 'autofs' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `autofs'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `autofs'
initctl: Unknown job: urandom
insserv: warning: script 'urandom' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `urandom'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `urandom'
initctl: Unknown job: K01nfs-common
insserv: warning: script 'K01nfs-common' missing LSB tags and overrides
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S01dbus
insserv: warning: script 'S01dbus' missing LSB tags and overrides
initctl: Unknown job: S01dbus
initctl: Unknown job: S01dbus
initctl: Unknown job: S01dbus
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S03nfs-common
initctl: Unknown job: S03checkroot-bootclean.sh
insserv: warning: script 'S03checkroot-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S03mountnfs-bootclean.sh
insserv: warning: script 'S03mountnfs-bootclean.sh' missing LSB tags and overrides
insserv: Service udev has to be enabled to start service udev-finish
insserv: exiting now!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package udev (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 udev
E: Sub-process /usr/bin/dpkg returned an error code (1)`

mkreisl, 2016-04-05:
Please show me the output of 
```
ls -la /etc/rcS.d/*udev*
```

er0l, 2016-04-05:
`lrwxrwxrwx 1 root root 21 Feb 18 07:31 /etc/rcS.d/S07udev-finish -> ../init.d/udev-finish`

mkreisl, 2016-04-05:
Ok,  ugly workaround:

```
cd /etc/rcS.d
sudo ln -s ../init.d/udev S01udev
```
How old is your installation? 
All my (old) installations having this link. But on a fresh installed RPi3 this link does not exist. Have to investigate later

er0l, 2016-04-05:
Thanks! Workaround proved to be adequate. 

Installation is from RPi2 which was always up-to-date, which I have updated with necessary packages to work with RPi3.

I have a lot of useful things configured so I did not use fresh installation. Are there any other differences that should be taken care of when using old RPi2 installation?

mkreisl, 2016-04-05:
> I have a lot of useful things configured so I did not use fresh installation. Are there any other differences that should be taken care of when using old RPi2 installation?

Don't know. I'm wondering why this link is sometimes there and sometimes not. Just looked into a very old RPi1 installation, there s NO link and having same issue

er0l, 2016-04-05:
Although I have been long time user even on my first RPi1 I believe that I have used fresh install when I switched to RPi2. Since then I have configured Xbian for other purposes and I was lazy to fresh install when switched to RPi3. There was just too many things to be copied :) Thanks anyway for your excellent work and support.

ksinitsyn, 2016-04-09:
I have same issue after update. Unfortunately workaround with link didn't help.
The biggest problem is that this device is located far away and I cannot reinstall xbian on it(

`root@xbian / # apt-get upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Create a snapshot of '/run/btrfs-snap/root/@' in '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-1958'
Delete subvolume (no-commit): '/run/btrfs-snap/root/@last_good_known'
Create a snapshot of '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-1958' in '/run/btrfs-snap/root/@last_good_known'
Delete subvolume (no-commit): '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-1856'
@btrfs-auto-snap_apt-run-2016-04-09-1958, 1 created snapshots, 1 destroyed snapshots, 0 warnings.
Setting up lvm2 (2.02.111-2.2) ...
[ ok ] Setting up LVM Volume Groups...done.
update-initramfs: deferring update (trigger activated)
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
initctl: Unknown job: K01nfs-common
insserv: warning: script 'K01nfs-common' missing LSB tags and overrides
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S02dbus
insserv: warning: script 'S02dbus' missing LSB tags and overrides
initctl: Unknown job: S02dbus
initctl: Unknown job: S02dbus
initctl: Unknown job: S02dbus
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S03nfs-common
initctl: Unknown job: S03checkroot-bootclean.sh
insserv: warning: script 'S03checkroot-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S03udev-finish
insserv: warning: script 'S03udev-finish' missing LSB tags and overrides
initctl: Unknown job: S03mountnfs-bootclean.sh
insserv: warning: script 'S03mountnfs-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S01udev
insserv: warning: script 'S01udev' missing LSB tags and overrides
insserv: Service mountdevsubfs has to be enabled to start service lvm2
insserv: exiting now!
update-rc.d: error: insserv rejected the script header
dpkg: error processing package lvm2 (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for initramfs-tools (0.120+deb8u1) ...
update-initramfs: Not updating initramfs.
Errors were encountered while processing:
 lvm2
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

mkreisl, 2016-04-09:
@ksinitsyn 
please show me `ls -la /etc/init.d/m*` should look like this
```
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/motd -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountall-bootclean.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountall.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountdevsubfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountkernfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountnfs-bootclean.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mountnfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Mär 26 21:03 /etc/init.d/mtab.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr 20  2015 /etc/init.d/mtab.sh.dpkg-old -> /lib/init/upstart-job

```

if link  `/etc/init.d/mountdevsubfs.sh -> /lib/init/upstart-job` does not exist, please add it manually by `sudo ln -s /lib/init/upstart-job /etc/init.d/mountdevsubfs.sh` and try again

ksinitsyn, 2016-04-09:
Everything looks same here.

```
root@xbian ~ # ls -la /etc/init.d/m*
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/motd -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountall-bootclean.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountall.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountdevsubfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountkernfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountnfs-bootclean.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mountnfs.sh -> /lib/init/upstart-job
lrwxrwxrwx 1 root root 21 Apr  6 08:49 /etc/init.d/mtab.sh -> /lib/init/upstart-job
```

mkreisl, 2016-04-09:
next ugly workaround
```
cd /etc/rcS.d
sudo ln -s /etc/init.d/mountdevsubfs.sh S01mountdevsubfs
```

please note: NO .sh suffix at S01mountdevsubfs

ksinitsyn, 2016-04-09:
Looks like passed!! Thanks!
Though I'm bit concerned about all this warnings (Unknown job)

```
root@xbian /etc/rcS.d # apt-get upgrade 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Create a snapshot of '/run/btrfs-snap/root/@' in '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-2114'
Delete subvolume (no-commit): '/run/btrfs-snap/root/@last_good_known'
Create a snapshot of '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-2114' in '/run/btrfs-snap/root/@last_good_known'
Delete subvolume (no-commit): '/run/btrfs-snap/root/@btrfs-auto-snap_apt-run-2016-04-09-1900'
@btrfs-auto-snap_apt-run-2016-04-09-2114, 1 created snapshots, 1 destroyed snapshots, 0 warnings.
Setting up lvm2 (2.02.111-2.2) ...
[ ok ] Setting up LVM Volume Groups...done.
update-initramfs: deferring update (trigger activated)
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
initctl: Unknown job: K01nfs-common
insserv: warning: script 'K01nfs-common' missing LSB tags and overrides
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S02dbus
insserv: warning: script 'S02dbus' missing LSB tags and overrides
initctl: Unknown job: S02dbus
initctl: Unknown job: S02dbus
initctl: Unknown job: S02dbus
initctl: Unknown job: K01nfs-common
initctl: Unknown job: S03nfs-common
initctl: Unknown job: S03checkroot-bootclean.sh
insserv: warning: script 'S03checkroot-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S03udev-finish
insserv: warning: script 'S03udev-finish' missing LSB tags and overrides
initctl: Unknown job: S03mountnfs-bootclean.sh
insserv: warning: script 'S03mountnfs-bootclean.sh' missing LSB tags and overrides
initctl: Unknown job: S01udev
insserv: warning: script 'S01udev' missing LSB tags and overrides
initctl: Unknown job: S01mountdevsubfs
insserv: warning: script 'S01mountdevsubfs' missing LSB tags and overrides
initctl: Unknown job: nfs-common
insserv: warning: script 'nfs-common' missing LSB tags and overrides
initctl: Unknown job: bootmisc.sh
insserv: warning: script 'bootmisc.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `bootmisc.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `bootmisc.sh'
initctl: Unknown job: checkroot-bootclean.sh
insserv: warning: script 'checkroot-bootclean.sh' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `checkroot-bootclean.sh'
initctl: Unknown job: mountall-bootclean.sh
insserv: warning: script 'mountall-bootclean.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountall-bootclean.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountall-bootclean.sh'
initctl: Unknown job: mountall.sh
insserv: warning: script 'mountall.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountall.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountall.sh'
initctl: Unknown job: mountnfs-bootclean.sh
insserv: warning: script 'mountnfs-bootclean.sh' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountnfs-bootclean.sh'
initctl: Unknown job: mountnfs.sh
insserv: warning: script 'mountnfs.sh' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `mountnfs.sh'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `mountnfs.sh'
insserv: warning: script 'dbus' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `dbus'
initctl: Unknown job: mtab.sh
insserv: warning: script 'mtab.sh' missing LSB tags and overrides
[... 45 lines cut ...]
insserv: warning: script 'procps' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `procps'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `procps'
insserv: warning: script 'udev' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `udev'
initctl: Unknown job: udev-finish
insserv: warning: script 'udev-finish' missing LSB tags and overrides
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `udev-finish'
initctl: Unknown job: fake-hwclock
insserv: warning: script 'fake-hwclock' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `fake-hwclock'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `fake-hwclock'
initctl: Unknown job: udev-mtab
insserv: warning: script 'udev-mtab' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `udev-mtab'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `udev-mtab'
insserv: warning: script 'autofs' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `autofs'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `autofs'
initctl: Unknown job: urandom
insserv: warning: script 'urandom' missing LSB tags and overrides
insserv: Default-Start undefined, assuming empty start runlevel(s) for script `urandom'
insserv: Default-Stop  undefined, assuming empty stop  runlevel(s) for script `urandom'
```
I installed lvm2 on another device with XBian and didn't get any of these warnings.

mkreisl, 2016-04-09:
Agree. Makes me nervous too. All these complained services are handled by upstart.

ksinitsyn, 2016-04-09:
Anyway. Thanks for help!

coderpop, 2016-04-10:
I have seen these warnings too on more than one pi.... But it doesn't seem to make a problem for Xbain as all works as expected even after this.

closed 2016-06-17 by mkreisl