xbianonpi/xbian #819: Under volt rainbow square - RPi 3

rikardo1979, 2016-04-24:
Trying to figure out why I have rainbow square indicating undervoltage?
Using exactly same PSU and USB cables as it was on RPi 2 . PSU rated at 2.4A and never had issue with it. Same with 10000mA power bank. 
Strange behaviour as I think is only with XBian

mkreisl, 2016-04-24:
Don't have this rainbow square. Even with old 5V/1A power supply and Ethernet + WLAN attached it does not appear.

(kernel 4.4.7, kodi 16.1 rc3, 2 weeks old installation)

rikardo1979, 2016-04-24:
older build here, RC2. Trying update now from within xbian-config in GUI

rikardo1979, 2016-04-24:
ok, seems I can not upgrade to RC3. will have to find some time to reflash and see if is going to be better

rikardo1979, 2016-04-24:
so upgrade did not help as you can see 
https://snag.gy/UWuFy4.jpg

bairdy, 2016-04-24:
@rikardo1979 
are you using wireless as i have the same issue with a 2A power supply from modmypi when its under a bit of load (updating xbian/cleaning library) and on wireless, but when using Ethernet it works fine with no issues and fine with my pi2. also your powerbank may say 10000mA but thats just the battery capacity it will more than likely have 5V/1A or 5V/2A stamped some where for the actual output rating but depending on it's quality may output slightly less.

rikardo1979, 2016-04-24:
nope, no wifi here.never. Always cable for every single device in my household :)
PSU is rated 2.4A and another at 2.1A and that is stable +5V as I have tested with DROK USB meter. 
I have used power bank for test purpose only and that is solid made unit with 2 USB outs, each rated 2A and yet again tested with meter. There is no drop under +5V for even split of the second.
There is nothing plugged into RPi apart from LAN and HDMI. 
SD card is solid Samsung 16gb, almost new.

rikardo1979, 2016-04-24:
and is even more funny with this latest build that I have no sound as extra bonus :)
There is this red mute icon in top right corner no matter what settings for audio I use.
My normal settings are HDMI out at 5.1 and all pass through with DTS/DD capable receiver.
But now I have this icon and no sound

bairdy, 2016-04-24:
have you tried changing kernel to a lower version as i've just updated to 4.4.7 and now suffering the rainbow idol  with just xbmc running,  a downgrade back to 4.4.6+-1458753791 seems to fix it for me. as for audio have you tried loading kodi web interface and pressing the mute/volume keys to get it back?

rikardo1979, 2016-04-24:
@bairdy thx for suggestions 
A bit to late now as I just reflashing img again :D but if this going to persist I will try those ;)

rikardo1979, 2016-04-24:
ok.sound seems to be fixed but the rainbow square undervolt indicator is still there...
No good but will live with it till fixed

mkreisl, 2016-04-24:
@rikardo1979 Try this: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=82373

Add `avoid_warnings=1` to /boot/config.txt

IMO sounds you have hardware failure.

rikardo1979, 2016-04-25:
I will test other distros again as I do not think is HW failure

rikardo1979, 2016-04-25:
OK, I was wrong :( its same with OL...I dont get it.
Will try return new RPi3 and get replacement.
Thank you and sorry for mess

closed 2016-04-25 by rikardo1979

popcornmix, 2016-04-25:
Pi3 needs more power than pi2. Your problem is 100% the power supply.
A power supply like:
https://shop.pimoroni.com/products/raspberry-pi-universal-power-supply
won't result in an under voltage condition.

rikardo1979, 2016-04-25:
@popcornmix well, I have tried more than 3 as I mentioned. The latest was 4A!
Same issue

popcornmix, 2016-04-25:
Note the cable from power supply is just as important as the supply itself.

rikardo1979, 2016-04-25:
@popcornmix as stated, I have tried multiple USB cables. At least 4 different cables and 4 different PSU units with measured and confirmed output current over 2A

rikardo1979, 2016-04-25:
i have requested RMA from CPC on this for money back

mkreisl, 2016-04-25:
IMO power supply is very critical and seems to be still buggy (you remember Xenon flashlight issue with RPi2). Another user on IRC had similar issue (rainbow square) after updating some packages. I suppose his RPi3 died also.

rikardo1979, 2016-04-25:
RMA done. replacement on the way. Will see.
It would be really odd if this shows up as PSU issue. I have tried 4 different ones.
And will not spend any more money for sure. If is going to be same I throw it to bin and get good old RPi2 or something else

popcornmix, 2016-04-25:
A voltmeter on the 5V line of GPIO header would be the best way to find out for sure.

rikardo1979, 2016-04-25:
@popcornmix can do once I receive another one. But DROK meter showed stable +5V with no drops below what so ever

rikardo1979, 2016-04-25:
Only an old version of RPi but here is voltage on GPIO pin
https://snag.gy/WYFwsk.jpg

PS:sorry for my old multimeter with damaged display :D thats 5.05V on ;)

popcornmix, 2016-04-25:
Make sure you measure when under load (e.g Kodi is running and busy, perhaps scrolling through fanart).

rikardo1979, 2016-04-25:
no matter if is busy or idle doing nothing the under volt was constantly flashing from 1st second it starts booting and than never stops even the static GUI was on screen.