xbianonpi/xbian #821: Mount root volume in read only

belese, 2016-05-07:
Hello All, 

It's not really an issue, but i'm doing a project with XBian, and it is battery powered, 
as i don't find any product with some stock to use battery and i2c to communicate with pi, 
i'm using a standard battery charger (like usb bank battery), 
problem is i can't shutdown the pi properly (it's a wood case with only 3 buttons for a kid).
So i would like to mount the root volume in read only, 
before i investigate a bit more, is someone know if i could run into problem by doing that?

My idea was to add a parameter in command_line to mount the root volume in readonly, 
i already look a bit at the code, i imagine it's in the initramfs i've to do it.

So if you have some advice before i start, thanks!.

Belese

mkreisl, 2016-05-07:
IMO it's not good idea mounting root ro. Gave a short test, system won't boot.

vondrt4, 2016-05-07:
What you need is some live CD kung-fu. The idea is to mount the flash card read only, but give the system write access to all areas it needs. The quick way is to mount an unionfs, mhddfs, or whatever is popular these days - probably aufs - OVER the read-only filesystem. This gives you read-write access everywhere, the modifications go to a ramdisk, zram, dedicated partition..

belese, 2016-05-07:
ok thanks for your answer and the test. 
i don't want to do an overlay, i don't have skill and i will wait if card is often corrupt, for now, it was never corrupt during my test. hope and see! i will close it as it's not an issue.

closed 2016-05-07 by belese