xbianonpi/xbian #823: RPi 3 Kodi 16.1 fails on Composite output

inkicthus, 2016-05-18:
Looks very similar to #796, but I'm on 16.1 (Git:2016-04-28-195c90b). When running on HDMI, everything works beautifully. When I switch to composite output (to connect to an old SDTV) kodi fails to start. I see the "Starting Kodi" splash image, then it goes black for a couple minutes and drops out to a terminal login prompt.

The [log file](http://pastebin.com/yUSPZmge) doesn't show any obvious errors, it just seems to stop suddenly while looking for a DVD (which my pi does not have). I also have the same problem on an old Rpi1 B+ that _was_ working back on 16.0 but since upgrading to 16.1 it doesn't output composite anymore either. I can get a log file from there too if that will help. Thanks!

mkreisl, 2016-05-18:
@inkicthus 
Please look here https://github.com/xbianonpi/xbian/issues/812#issuecomment-218714192

inkicthus, 2016-05-18:
Thanks for the quick reply, sorry I didn't recognize that issue as being connected. [This package](http://apt.xbian.org/pool/devel/rpi2-jessie/x/xbian-package-xbmc/xbian-package-xbmc_16.1-1462731607_armhf.deb) appears to work. At least kodi starts up and I can see it, haven't done much real testing yet.

Do I need to worry about that being overwritten in by any other updates? I have automatic updates turned off for now. Thanks again!

mkreisl, 2016-05-18:
> Do I need to worry about that being overwritten in by any other updates? I have automatic updates turned off for now. Thanks again!

Maybe good idea to set package xbian-packgage-xbmc on hold. I can't promise that this issue will be fixed in 16.x, hopefully it will be work in 17.x again

inkicthus, 2016-05-20:
Got it. Thanks again!

closed 2016-05-20 by inkicthus

mkreisl, 2016-05-23:
> Maybe good idea to set package xbian-packgage-xbmc on hold. I can't promise that this issue will be fixed in 16.x, hopefully it will be work in 17.x again

Good news: it has been solved for 16.x version. New version is now available in stable repo

inkicthus, 2016-05-31:
Thanks for the update! Confirmed working on stable version.