xbianonpi/xbian #824: Update Super Repo URL

NunoFilipeSantos, 2016-05-22:
I am under the impression that the [Super Repo](https://superrepo.org) that comes bundles in Repositories section and File manager has an out of date URL.

The new URL that is [supposed to go there](https://superrepo.org/get-started/) is http://srp.nu.

This is a minor nuisance, since it is easy to change it manually, but having the updated one would be fantastic.

mkreisl, 2016-06-17:
Don't know where it came from. 
Grepped complete source tree of Kodi, but there is nothing there.
So I would suppose that Kodi fetches those wrong data from somewhere.

mkreisl, 2016-07-06:
Coincidentally found that place where it is done :smile:

Fix is WIP now

mkreisl, 2016-07-06:
Fixed

closed 2016-07-06 by mkreisl