xbianonpi/xbian #827: Devel and kodi 17a3

coderpop, 2016-08-07:
Hi,

Had a little time today so tried this again. Audio is now working on I2S with some dropouts. I think it's related, but the 'lost connection to server' message is flashing up most of the time on the http page.
I run with SMB mounts via network to a couple of NAS boxes with media on.

Hope this might be of some use.

mkreisl, 2016-08-07:
> Had a little time today so tried this again. Audio is now working on I2S with some dropouts. I think it's related, but the 'lost connection to server' message is flashing up most of the time on the http page.

You're talking in riddles. what is meant by `http page` and where do you see the `lost connection message` ???

coderpop, 2016-08-07:
I mean when you connect to the pi via briwser to the pi with http and get the kodi page.
Sorry if it wasn't clear.

coderpop, 2016-08-22:
Today I tried again to go to Kodi 17 on a Pi2 and this time it seems to be a success. I only tried with headless pi so far, but have good I2S sound and the web interface is working (although in the intrim I changed from default to yarc, but I doubt it made much difference). This is on a stereo audio DAC.

Now I have also tried a Pi2 with toslink passthrough device and it doesn't give any sound. I have looked into it and on the system>settings>audio page the correct hifiberry-digi is selected, under audio passthrough the option to select passthrough device is greyed out and says PI:HDMI. 

This is probably a Kodi issue, but there is now no option to select 50Hz refresh rate for the video output. Also, 'play next track automatically' does not work for music. Again I expect this to be a Kodi WiP issue.

mkreisl, 2016-08-22:
> So now I have also tried a Pi2 with toslink passthrough device and it doesn't give any sound. I have looked into it and on the system>settings>audio page the correct hifiberry-digi is selected, under audio passthrough the option to select passthrough device is greyed out and says PI:HDMI. This is probably the issue.

This patch https://github.com/xbianonpi/xbian-sources-xbmc/commit/28924ec14bd9779589b49ca74a5c6bc547e2f828 is currently not applied
But I'll keep this in mind :smile: 

> These two are probably a Kodi issue, but there is now no option to select 50Hz refresh rate for the video output. Also, the music 'play next track automatically' feature does not work for me. Again I expect this to be a Kodi WiP issue.

Which `two` ?

I can select 1080p/50Hz output, at least with Beta1.

Yes, probably your music play issue is a Kodi WIP issue

closed 2017-09-07 by mkreisl