xbianonpi/xbian #829: xbian-package-transmission doesn't auto start at boot

jdeluyck, 2016-09-07:
After having enabled transmisison through kodi, xbian-package-transmission version 2.82 got installed.
The package seems to have a bug in the init script, though, because of which it never gets started:

# Required-Start:    networking
# Required-Stop:     networking

this should read:
# Required-Start:    $network
# Required-Stop:     $network

mkreisl, 2016-09-07:
@jdeluyck 
Built completely new package, has version 2.92 now and `/etc/init.d/transmission` has been fixed
Could you please test it (please note: currently only available for Rpi2/3 and visable at midnight (mirrors have do sync))

jdeluyck, 2016-09-08:
@mkreisl works for me (TM)

closed 2016-11-04 by mkreisl