xbianonpi/xbian #830: kodi fails to start at boot when /home is moved to external hdd

jdeluyck, 2016-09-07:
RPI2, wanted to move /home to external hdd because of the 1. excessive writes kodi was giving to the SD, and 2. for a speed boost. 

Logged in as root, killed all services, changed /etc/fstab, tar'ed /home/xbian and moved it to the external hdd partition. All works, except kodi won't start at boot.

SSH'ing in and executing 'start xbmc' starts kodi without a problem, so I'm thinking it might be a dependency issue somewhere, but I'm not finding it. /var/log/upstart/ logs aren't exactly useful to find why kodi fails. 

There is no log created under /home/xbian/.kodi/ either.

mkreisl, 2016-09-07:
I suppose xbmc job is started too early (before all mounts are done)

Do you see the Kodi start splash screen?

jdeluyck, 2016-09-07:
Do you mean the Xbian "Starting Kodi" screen? Yeah. Eventually it times out and bombs me back to a login prompt.

mkreisl, 2016-09-08:
@jdeluyck 
Investigated this issue and found solution

1) make sure that your entry for /home in fstab looks like this:
`/dev/sda1             /home                   xbian   subvol=xbian,noatime,rw           0       0`
fstype must be xbian and option nobootwait has to be removed

2) In file /etc/init/xbmc-preload.conf:
change `start on started xbian-run` to `start on started xbian-run and filesystem`

In my test enviroment (external btrfs formatted 2.5" 1TB hd it takes up to 14s until /home is mounted). So, if option `nobootwait` is given, mount will be done in background and has been finished too late for Kodi (this is configuation issue). If `filesystem` is not present in file /etc/init/xbmc-preload.conf, Kodi is started before all mounts has been done (this is real issue)

Please can you test if this works in your environment so I can commit that change in file /etc/init/xbmc-preload.conf.

jdeluyck, 2016-09-13:
It works.. but does unfortunately require btrfs on the external disk. A requirement I can live with, as soon as I get another one to pump over my data ;)

mkreisl, 2016-09-13:
It does not require btrfs, my external disk is btrfs formatted and so it was easiest way to test by setting up a subvolume :smile:

closed 2016-11-04 by mkreisl