xbianonpi/xbian #831: xbiangit not set

CGarces, 2016-09-25:
I'm testing the build system of xBian, but i'm doing something wrong

I have cloned the repository and executed `./run.me prepare` with sudo

but   $xbiangit is not set... `echo  $xbiangit` not return anything.

mkreisl, 2016-09-25:
After first run you have to logout
Did you do this?

CGarces, 2016-09-25:
Yes, logout and login again

mkreisl, 2016-09-25:
Hmmm, did this on 4 different machines, was always successful

CGarces, 2016-09-25:
Strange...
Everithing looks fine

> Writing loader file /etc/profile.d/xbiangit.sh
> Installing 
> Installing done...
> Successfully initialised. Please logout and login again.

How is set xbiangit env variable?

I see 
`export xbiangit=$(pwd)/run.me` at create.buildroot
But nothing more.

mkreisl, 2016-09-25:
Please look into file `/etc/profile.d/xbiangit.sh`

CGarces, 2016-09-25:
Fixed, was a weird issue with my path

> $ echo $PATH
> .:.:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

Removing the dots at the beginning fix the issue.

closed 2016-09-25 by CGarces