xbianonpi/xbian #833: Online audio content streaming is poor - audio drops (iMX6)

Implexus, 2016-12-04:
Hi there,

Probably since you've switched to the RT kernels there is an issue with audio streams in Kodi. It drops audio for a very little time in random intervals (15-60s between each of them) for a few milliseconds. It's really a horrible experience to listen to the music like that.

My device is CuBox-i4 Pro (iMX6-Quad CPU) - nothing else is not running in the background.

Latest kernel working without the mentioned issues was version 4.3.3+ (linux-image-armmp=4.3.3+-1458345438).

Implexus, 2017-01-12:
The issue has been solved by @mk01

closed 2017-01-12 by Implexus

didlich, 2017-09-02:
@Implexus , could you paste the link to the solution please
after last update on my cubox I have the same problem

Implexus, 2017-09-04:
@didlich This bug-fix should be implemented already. Maybe there is another similar issue with the current release.

didlich, 2017-09-04:
OK, I'll create a new issue