xbianonpi/xbian #834: Crash on Open some Pictures with high resolution

Kudro75, 2016-12-07:
Hi freands,
my xbian on RP 2 crashes if i try to open some Pictures like this one
http://www.alan-sweets.com/test/DSC00050.JPG
Error message ist this:
21:23:39 102618.703125 T:1872368560   ERROR: COMXCoreComponent::DecoderEventHandler OMX.broadcom.image_decode - OMX_ErrorStreamCorrupt, Bitstream corrupt
21:23:39 102618.703125 T:1516237744   ERROR: COMXCoreTunel::Deestablish - Error WaitForCommand port 60 on component OMX.broadcom.image_decode omx_err(0x8000100b)

Can some one help

Regards an thanks for Help

f1vefour, 2016-12-08:
Are you opening them locally or over network? Is your Pi overclocked?

mkreisl, 2016-12-08:
I can confirm issue with your sample pic. 
Seems to be very oversized, if I edit the pic a bit (for example red eye correction but NO resolution change), new size is below 1MB and can be displayed correctly

Kudro75, 2016-12-09:
Dear mkreisl,
I have take all Pictures with my Sony "DSC-W730". I will not open and edit all images now :-).

Ich werde doch nicht alle Bilder jetzt öffnen und bearbeiten :-). Gibt es da euer seits eine andere Lösung.
Regards
Viele Grüße
Kudro

Kurdo75, 2016-12-09:
Dear f1vefour,
it is the same effect if opening them locally or over network.
my Pi2 is orginal image frob xbmc and i have not overclocked.
Regards

mkreisl, 2016-12-09:
@Kurdo75 

> I have take all Pictures with my Sony "DSC-W730". 

I know, Kodi already reporting this to me :smile: 

> I will not open and edit all images now :-).

Sure, it was my first thought that size is too big or resolution is too high, but it isn't.

> Ich werde doch nicht alle Bilder jetzt öffnen und bearbeiten :-). Gibt es da euer seits eine andere Lösung.

I'm working on it. Spent some hours yesterday and had little success, but still do not know what the culprit is. The problem is not displaying the picture, the problem is generating the thumbnails of them. Once thumbnails are there, everything is ok without any error messages in the log

Kurdo75, 2016-12-10:
thanks mkreisl,
I hope you find the solution 👍  :-)
Regards

mkreisl, 2016-12-10:
Yes, there is solution :smile: 

Usually, on RPi jpegs are rendered by GPU, but unfortunately the jpegs you have can not be rendered, no idea why.

Now option added to disable GPU rendering of jpegs. Build of new xbian-package-xbmc is in progress.

You'll find this option at same place where OMX and MMAL acceleration is, and in your case you have to disable it.

Kurdo75, 2016-12-12:
thanks very much for your great Help

mkreisl, 2016-12-12:
If your issue is solved, please close it here :smile:

Kurdo75, 2016-12-18:
it work. Thank you very much...great

May be a logical message in kodi may help user in future to find the reason...

By

closed 2016-12-18 by Kurdo75