xbianonpi/xbian #835: Xbian does not boot after the udpate.

pille5, 2016-12-28:
Hello,

I tried to update my Xbian, but there were some error during the update and now my raspberry does not boot. It fails with a kernel panic:

`Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block(179,2)`

How can I recover my system? I don't want to lose all my setup. Unfortunately I didn't do backup before the update. :(

I tried to do this: http://raspberrypi.stackexchange.com/questions/40854/kernel-panic-not-syncing-vfs-unable-to-mount-root-fs-on-unknown-block179-6/40855#40855

But seems to me the USB does not have any power during the boot sequence. I cannot use my keyboard, so cannot activate recovery command line. I read in multiple places, that I have to modify the `cmdline.txt`, but I only have `cmdline.new` `cmdline.txt.old` and `cmdline.default`. I changed `root=/dev/mmcblk0p2` to `root=/dev/mmcblk0p6` in every cmdline.* files, but didn't help. Please help.

mkreisl, 2016-12-28:
You need cmdline.txt. So you have to copy cmdline.txt.old to cmdline.txt and try again.

Are you running NOOBS on your RPi?

pille5, 2016-12-28:
@mkreisl, good question. :) As far as I remember I installed the latest Xbian with the Xbian installer. I think that does not contain NOOBS.

Anyway I managed to recover my system. I just found out there is an example `cmdline.txt` on the FAQ page with the default settings. I used that and it worked for me, so I close the issue. Sorry for the noise.

closed 2016-12-28 by pille5

mkreisl, 2016-12-28:
You managed it, that's fine.

I was asking because under normal installation root partition is always on /dev/mmcblk0p2, only under NOOBS environment partition is above /dev/mmcblk0p4