xbianonpi/xbian #836: Move to SPMC instead of kodi

CurlyMoo, 2017-01-11:
As you all might known @koying stopped developing for kodi. For as far as i know, the relationship between @mk01 and @koying was pretty good, whilst the relationship between XBian and kodi is discussable.

So, wouldn't it just be a good idea to start supporting SPMC and coorperating with @koying. Already running it myself on my Android TV (besides XBian) and it works really good.

f1vefour, 2017-01-11:
I also run SPMC on a couple of devices. But isn't SPMC code extremely different?

CurlyMoo, 2017-01-11:
It's just a fork of Kodi, but then with Android specific patches. And Kodi code doesn't have the best reputation after all...

mkreisl, 2017-01-11:
I don't think that this has future:

1) It is only for Android, so it is useless for XBian
2) Look at Github, there are 386 open issues, that does not really convincing me

CurlyMoo, 2017-01-11:
> It is only for Android, so it is useless for XBian

It is not *only* for android. As far as i know he doesn't *remove* code, he just adds android code that wasn't allowed anymore by the Kodi board members. It's maybe good to read about the background of @koying leaving.

TLDR; he was the android experts of XBMC but due to some (not quite rare cases) of disputes inside the XBMC team he left and decided to fully work on his fork.

So yes, the guy in a Android expert so he develops Android features (that Kodi doesn't have and probably not going to have).