xbianonpi/xbian #837: pvr recordings playback results in an immediate death

ssakatausa, 2017-03-26:
On RPi3, XBian 2017.03.24, XBian 2017.03.17, and XBian 2017.03.10 fail to play recorded TV programs through the pvr.mythtv addon, while XBian 2017.03.03 is not subject to the problem. To replicate the problem:

1. After fresh installation of any of the three versions of XBian, just set the IP address of the mythtv backend in the configuration screen of pvr.mythtv and enable pvr.mythtv.
2. Click any in the list of the recorded programs to start the video playback. 

This immediately results in the black screen. Kodi reacts to no keyboard input or mouse movement. Also, neither ssh access to the RPi3 not ping to the RPi3 works (the network seems to be down). The only thing I can do at this point is to unplug and re-plug the power cable to give a hard rest to the RPi3. Unfortunately, the log file contains no information about the problem, probably because the machine dies before any information about the recording playback is written to the log file.

By the way, I have already tried the latest version of pvr.mythtv (5.0.10) and found that it makes no difference. I hope that you can fix the problem. I will stick with XBian 2017.03.03 until then. Thanks!

mkreisl, 2017-03-26:
@ssakatausa 

Do not have test environment for mythtv, so can not figure out where the problem is.

Two major packages were updated (Kodi = **xbian-package-xbmc** and Kernel = **linux-image-bcm2836)** but none of those packages between Mar 03 and 10, only **cec** library has been updated on Mar 08.

So could you please update package for package and see if it still works?

Thanks

ssakatausa, 2017-03-27:
Thanks for your comment, mkreisl. I'll try what you suggested.

ssakatausa, 2017-03-31:
I am sorry that I could not try your suggestion earlier. Starting with a new installation of XBian 2017.03.03, I today upgrade all upgradable packages one by one. The upgradable packages were:

 btrfs-tools
 linux-image-bcm2836
 xbian-package-config-shell
 xbian-package-firmware
 xbian-package-repo
 xbian-package-xbmc
 xbian-update

I have found that the problem of pvr.mythtv recordings playback happens as soon as linux-image-bcm2836 is upgraded from 4.4.48+-1486919396 to 4.9.17+-1490545026. I have also verified that the upgrade of the packages does cause the problem if linux-image-bcm2836 is not upgraded. 

Best,

  S

mkreisl, 2017-04-02:
@ssakatausa 
Thanks for clarifying this. Now we know which package is the culprit.

I'm wondering why kernel 4.9 does not work in that case, because i'm using this version since months and never had issues, but using tvheadend and vdr vnsi pvr and not mythtv. Seems I have to reactivate my old mythv server

mkreisl, 2017-04-02:
So, mythtv back from retirement and can confirm that Kodi/Mythtv combination on RPi(3) is very very unstable when running kernel 4.9.
Sometimes I can can play live tv and recordings and sometimes not.

ssakatausa, 2017-04-02:
@mkreisl 
It is interesting that you can sometimes play live TV and recordings. In my experiments, Kodi instantly froze 100% times. By the way, I have found that kodi can play the recording video files on the mythtv backend server without any problems via the video menu. It is very puzzling that accessing the same file via pvr.mythtv gets kodi frozen.

mkreisl, 2017-04-07:
@ssakatausa 

Could you please turn off dynamic priority (can be done in xbian-config GUI or manually in file /etc/default/xbmc, DPRIORITY=no) and check again kernel 4.9

Using dynamic priority is culprit (not sure what happens exactly, if kernel having issue or whatever)

ssakatausa, 2017-04-07:
@mkreisl 

Yes, turning off dynamic priority fixes the problem under kernel 4.9.

mkreisl, 2017-04-08:
Dynamic priority disabled per default now

> But it also makes HD playback a little sluggish compared to that with dynamic priority "on" under the previous kernel

Can't see any difference here

mkreisl, 2017-04-08:
Btw, running kernel 4.10.8 on my i.MX6 Hummingboard now. 
Will test if this issue occurs when enabling dynamic priority