xbianonpi/xbian #838: libcec3 is not removed when libcec4 is installed

ssakatausa, 2017-03-27:
I have just noticed that both libcec3 and libcec4 are installed on my RPi3. Isn't libcec4 supposed to replace libcec3? On my machine, "dpkg -l libcec\*" shows:

Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                               Version                Architecture           Description
+++-==================================-======================-======================-==========================================================================
ii  libcec                             4.0.2.1~xbian-14889991 armhf                  Meta package libCEC.
ii  libcec-dev                         4.0.2.1~xbian-14889991 armhf                  libCEC communication Library (development files)
un  libcec1                            <none>                 <none>                 (no description available)
un  libcec2                            <none>                 <none>                 (no description available)
ii  libcec3:armhf                      3.1.0.1~xbian-14628598 armhf                  libCEC communication Library (shared library)
ii  libcec4:armhf                      4.0.2.1~xbian-14889991 armhf                  libCEC communication Library (shared library)

Best.

mkreisl, 2017-03-27:
That's right, libcec3 is not removed when libcec4 is being installed, but this does not matter because of different .so files. You can remove libcec3 manually.
Due to an minor issue in package list when building image libcec3 and libcec4 were both installed. This has been fixed at March 7, see this commit https://github.com/xbianonpi/xbian/commit/ba1003aa312a37397ec44c7ea69f0d33ca70aa04

ssakatausa, 2017-03-27:
Thanks for your reply. I now see what is going on.

closed 2017-03-27 by ssakatausa