xbianonpi/xbian #840: Boot problem

pcccpcccp, 2017-04-28:
Hi, 
Since last update (V17 & 17.1) i encounter several stability problem, some time my screen stay black and my only possibility is to reboot my Pie.
Since this morning i boot without X, 
When i check the kodi status it send me 
`xbmc stop/waiting`
I already try to do proper reboot, electrical reboot, kodi start.
Nothing work...

mkreisl, 2017-04-28:
Log files?
Maybe hardware failure? Already tried reinstalling xbian-package-xbmc?
Btw, we do not run X

pcccpcccp, 2017-04-28:
what log do you want  ?
No hardware work fine because some service start (for exemple i can reach my ftp server or my transmission deamon.) Only GUI is not available...
No i don't try to reinstalle xbian-package-xbmc

mkreisl, 2017-04-28:
apt-get install --reinstall xbian-package-xbmc

Kodi log in /home/xbian/.kodi/temp

pcccpcccp, 2017-04-28:
thks i'm trying to reinstall xbmc

Kodi.log`14:05:17.796  36.016911 T:1954653008  NOTICE: GL_EXTENSIONS = GL_OES_compressed_ETC1_RGB8_texture GL_OES_compressed_paletted_texture GL_OES_texture_npot GL_OES_depth24 GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_EGL_image_external GL_EXT_discard_framebuffer GL_OES_rgb8_rgba8 GL_OES_depth32 GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_APPLE_rgb_422 GL_EXT_debug_marker
14:05:19.868  38.088856 T:1817666480  NOTICE: Register - new cec device registered on cec->RPI: CEC Adapter (2708:1001)
14:05:19.871  38.092148 T:1925948336  NOTICE: Running database version Addons27
14:05:19.882  38.103313 T:1925948336  NOTICE: Running database version ViewModes6
14:05:19.890  38.111729 T:1925948336  NOTICE: Running database version Textures13
14:05:19.907  38.128517 T:1925948336  NOTICE: Running database version MyMusic60
14:05:19.940  38.161079 T:1925948336  NOTICE: Running database version MyVideos107
14:05:19.963  38.184563 T:1925948336  NOTICE: Running database version TV29
14:05:19.972  38.193218 T:1925948336  NOTICE: Running database version Epg11
14:05:20.566  38.787315 T:1954653008  NOTICE: start dvd mediatype detection
`
Kodi.old.log
`12:05:04.124  22.336578 T:1954349904  NOTICE: Default Video Player: VideoPlayer
12:05:04.124  22.336786 T:1954349904  NOTICE: Default Audio Player: paplayer
12:05:04.124  22.336906 T:1954349904  NOTICE: Disabled debug logging due to GUI setting. Level 0.
12:05:04.124  22.337009 T:1954349904  NOTICE: Log level changed to "LOG_LEVEL_NORMAL"
12:05:04.128  22.340569 T:1954349904  NOTICE: CMediaSourceSettings: loading media sources from special://masterprofile/sources.xml
12:05:04.132  22.344423 T:1954349904  NOTICE: Loading player core factory settings from special://xbmc/system/playercorefactory.xml.
12:05:04.138  22.350893 T:1954349904  NOTICE: Loaded playercorefactory configuration
12:05:04.139  22.351967 T:1954349904  NOTICE: Loading player core factory settings from special://masterprofile/playercorefactory.xml.
12:05:04.140  22.352654 T:1954349904  NOTICE: special://masterprofile/playercorefactory.xml does not exist. Skipping.
14:05:04.322  22.535055 T:1954349904  NOTICE: Running database version Addons27
`

mkreisl, 2017-04-28:
Sorry, but log is absolutely UNREADABLE.

Please remove kodi.log and kod.old.log before starting Kodi and then please pastebin log and post link here

pcccpcccp, 2017-04-28:
Hum worst than ever...
after reinstalling same problem but now there no more log...(no more kodi.log or kodi.old.log)
and now after sending a reboot command i need to do electrical reboot...

mkreisl, 2017-04-28:
Check your filesystem and/or check if your service is enabled:

```
sudo btrfs scrub start /;sudo watch btrfs scrub status /

sudo cat /etc/init/xbmc.override
```

pcccpcccp, 2017-04-28:
seems i'have 220 uncorectable errors... scrubs is still running, I assume that i have to reinstall everything 

about xbmc.override 
`cat: /etc/init/xbmc.override: No such file or directory`

mkreisl, 2017-04-28:
If you have Samsung sd-card, your card is probably broken (card can't be written, but does not reporting this)

pcccpcccp, 2017-04-28:
no it's not a samsung card (i don't really like samsung) it's a sandisk. btw you can close this ticket un less you have a magical solution ;)

mkreisl, 2017-04-28:
Ok, probably reinstalling is faster than repairing fs :smile:

closed 2017-04-28 by mkreisl