xbianonpi/xbian #841: Pi Zero W running underclocked

aladar42, 2017-05-12:
For some reason Xbian runs Pi Zero W way underclocked - after installing I noticed a huge slowdown compared to LibreELEC, and found out that Xbian is running ARM at 700Mhz and core at 250, while the native clock should be 1000/400. Overclocking in xbian-config suggested 850Mhz ARM and then quit with an error. Is the system installing a profile optimized for RPi 1?

mkreisl, 2017-05-12:
Not sure what you did, but the overclocking profiles has been not updated for a long time

Please look at your /boot/config.txt which values are currently used

aladar42, 2017-05-12:
The 700/250 were the default values, that's why I tried to go to the overclocking menu. I don't have the system installed at the moment, I'll try a fresh install tomorrow.

mkreisl, 2017-05-12:
Ok, so please post output when you're ready 😄 

I'll try to fix that function asap

aladar42, 2017-05-13:
So, I just flashed a fresh image (12.5.2017).

xbian@xbian ~ $ vcgencmd measure_clock arm
frequency(45)=700000000
xbian@xbian ~ $ vcgencmd measure_clock core
frequency(1)=250000000

Just so I could make sure it wasn't me screwing something up.

The outputs:

```
xbian@xbian ~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 835.58
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000005bd62a5a
```

```
xbian@xbian ~ $ rpi_revision
memory: 512 MB
processor: Broadcom BCM2835
i2cDevice: /dev/i2c-1
model: Model Zero W
manufacturer: Sony
pcb revision: 1
warranty void: no
revision: 9000c1
peripheral base: 0x20000000

checking peripheral base against bcm_host library
... peripheral base matches base from bcm_host

```

(hope I didn't mess up the styling of this, I'm not familiar with github styling.

mkreisl, 2017-05-13:
Your styling is perfect 👍 
Output is what I expected

I'm wondering about 

> xbian@xbian ~ $ vcgencmd measure_clock arm
> frequency(45)=700000000
> xbian@xbian ~ $ vcgencmd measure_clock core
> frequency(1)=250000000

this values. Did you try overclock function again?

Or is governor set to ondemand?

```
xbian@kmxbilr:~$ sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
ondemand
```

aladar42, 2017-05-13:
Nope, I didn't touch anything. That's how it runs from a fresh install. The only thing I did was to set it up to connect to my Wifi. 
I think it's trying to pull a RPi1 values or something.

```
xbian@xbian ~ $ sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
performance
```

mkreisl, 2017-05-13:
Hmmm, thats weird.

Could you please show me /boot/config.txt, to make sure that there are no overclock values set

(performance is that what I was expecting)

Thanks

aladar42, 2017-05-13:
```
#dtoverlay=pi3-disable-bt
arm_freq=840
core_freq=320
sdram_freq=400
```

mkreisl, 2017-05-13:
Strange, no idea where that *_freq parameters came from. Could you please delete them and look if they are back when you reboot Pi?

Anyway, now I have all informations needed for fixing the overclocking module if xbian-config

When finished, there will be at least new package `xbian-package-config-shell`

aladar42, 2017-05-13:
Deleted the values, didn't come back. Seems like the correct values are in place now?

```
xbian@xbian ~ $ vcgencmd measure_clock arm
frequency(45)=1000106000
xbian@xbian ~ $ vcgencmd measure_clock core
frequency(1)=399999000
```

So, I'm not exactly a coder, but am I right in that XBian detects the device wrongly as Pi 1 during first boot or installation? Because those values were there from the start, I didn't even touch xbian-config.

mkreisl, 2017-05-13:
> So, I'm not exactly a coder, but am I right in that XBian detects the device wrongly as Pi 1 during first boot or installation? Because those values were there from the start, I didn't even touch xbian-config.

Yeah, seems that initial boot is doing some strange things. But could not found place there this has been done

aladar42, 2017-05-14:
Well, the problem still persists. Even after having the right clock set up, I'm not able to play most x264 720p files. It will just freeze up every few seconds. Seems like XBian just doesn't work on Zero W.

mkreisl, 2017-05-14:
Probably you have WLAN issue, the onboard WLAN adapter is really not reliable

mkreisl, 2017-05-14:
I looked for special settings needed for Zero W, but there is no need to change anything. Standard RPi1 kernel and Kodi build for RPi1 has to work flawlessly on 0W, and that overclock issue I have already figured out where this came from.

aladar42, 2017-05-14:
It's definitely not WLAN issue, because LibreELEC is running all of the videos flawlessly...