xbianonpi/xbian #842: OpenVPN quits due to nonexisting /dev/net/tun

aladar42, 2017-05-13:
I am running the latest version of XBian, and running OpenVPN fails.

```
Sat May 13 13:42:01 2017 ERROR: Cannot open TUN/TAP dev /dev/net/tun: No such file or directory (errno=2)
Sat May 13 13:42:01 2017 Exiting due to fatal error
```

There are several "fixes" that I found on the internet (running modprobe tun, etc), that do work, but only until restart. After restarting the device, OpenVPN fails to start again. Manually creating the /dev/net/tun dir also works only until reboot. Any idea what to do?

mkreisl, 2017-05-13:
Please add tun to /etc/modprobe (don`t forget to add <return> after tun :smile:)

```
cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

#8192cu
smsc95xx
#rtl8192cu
uinput
tun

```

aladar42, 2017-05-13:
/etc/modprobe didn't exist, so I created it.

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

uinput
tun
```

However,

```
xbian@xbian ~ $ cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

uinput
```

am I forgetting something? Is it possible to ship it already configured with the system?

mkreisl, 2017-05-13:
Ehm, sorry. Your Pi Zero W clock issue is confusing me

/etc/modprobe is not correct, I meant `/etc/modules` of course

aladar42, 2017-05-13:
Thanks, working now. Any chance to push it to the default modules file in the future?
Also, got another error, I'm assuming that's because XBian doesn't support ipv6? (Commenting out ipv6 tunnelling fixed it)

```
Sat May 13 15:34:24 2017 Linux ip -6 addr add failed: external program exited with error status: 2
Sat May 13 15:34:24 2017 Exiting due to fatal error

```

aladar42, 2017-05-13:
OK, sorry, false alarm, OpenVPN service still not starting automatically. I can only run it manually.

```
xbian@xbian ~ $ sudo service openvpn status
[FAIL] VPN 'mullvad_linux' is not running ... failed!
```

Not sure where the log file is.

mkreisl, 2017-05-13:
You have to install/enable syslog daemon, I prefer rsyslog

Seems that your openvpn service is not set to autostart. You can add openvpn as service in xbian-config (console and Kodi GUI version both works) and enable autostart

mkreisl, 2017-05-13:
XBian supports IPV6, but it's disabled per default

aladar42, 2017-05-13:
Thanks, looks like it starts now. I was used to desktop Debian adding the service to autostart automatically. :x However, it seems to start after Kodi initializes completely, wouldn't that be an issue?

mkreisl, 2017-05-13:
> Thanks, looks like it starts now. I was used to desktop Debian adding the service to autostart automatically. :x However, it seems to start after Kodi initializes completely, wouldn't that be an issue?

Don't think so. If you need openvpn up before Kodi starts, you can add openvpn to file /etc/default/xbmc, parameter WAITFOR

```
# Put name of all processes which XBMC should wait for before it will be started.
# e.g. WAITFOR=""
WAITFOR="openvpn"
```

aladar42, 2017-05-13:
Cool, all working perfectly now. Thanks. I'll close this issue, but I would still like to request if /etc/modules could come with tun already set up, so that people won't have to bother the devs like I did. :)

closed 2017-05-13 by aladar42