xbianonpi/xbian #843: Mediatek MT7601U USB bgn WiFi dongle not working

soxvpshubs, 2017-05-20:
I bought a XiaoMi WiFi dongle which works with Ubuntu but not working on Xbian.

Xbian unable to detect it but it is work on my Ubuntu PC.

```
xbian@xbian ~ $ lsusb
Bus 001 Device 004: ID 2717:4106  
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

```
xbian@xbian ~ $ uname -a
Linux xbian 4.9.28+ #1 PREEMPT Tue May 16 19:06:38 CEST 2017 armv6l GNU/Linux
```

I tried to install driver from [https://github.com/jamesfoley/raspberry-pi-MT7601](https://github.com/jamesfoley/raspberry-pi-MT7601) but it doesn't work.

Any idea to solve this problem?

mkreisl, 2017-05-21:
@soxvpshubs 

Please continue here: http://forum.xbian.org/thread-3120.html

mkreisl, 2017-10-31:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-10-31 by mkreisl