xbianonpi/xbian #846: Linux can't mount storage part of the SD card

aladar42, 2017-07-25:
Hey, I recently installed XBian again, and I can't access the storage part on the SD card. The system part mounts without a problem, but whenever I try to mount the storage partition, I get an error:

```
Failed to mount "xbian".
Error mounting /dev/sdc2 at /run/media/aladar/xbian: Wrong fs type, /dev/sdc2 has an invalid superblock or missing helper program..
```

btrfs device scan doesn't help. I'm running Manjaro on 4.12.2. btrfs-progs is installed.

```
$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0   2,7T  0 disk 
├─sda1   8:1    0   300M  0 part /boot/efi
├─sda2   8:2    0   2,7T  0 part /
└─sda3   8:3    0   8,8G  0 part [SWAP]
sdb      8:16   0 232,9G  0 disk 
├─sdb1   8:17   0   100M  0 part 
├─sdb2   8:18   0   128M  0 part 
└─sdb3   8:19   0 232,7G  0 part /run/media/aladar/D018DA6618DA4ADC
sdc      8:32   1    29G  0 disk 
├─sdc1   8:33   1    68M  0 part 
├─sdc2   8:34   1  28,7G  0 part 
└─sdc3   8:35   1   245M  0 part 
```

Is there anything I can do?

aladar42, 2017-07-25:
Tried mounting on another PC running Debian, same error. dmesg | tail gives:

```
[   84.012033] raid6: using intx1 recovery algorithm
[   84.032807] xor: measuring software checksum speed
[   84.072007]    prefetch64-sse:  6192.000 MB/sec
[   84.112007]    generic_sse:  6179.000 MB/sec
[   84.112008] xor: using function: prefetch64-sse (6192.000 MB/sec)
[   84.244503] Btrfs loaded, crc32c=crc32c-generic
[   84.246407] BTRFS: device label xbian devid 1 transid 964 /dev/sdb2
[   86.362582] BTRFS info (device sdb2): disk space caching is enabled
[   86.362592] BTRFS error (device sdb2): cannot mount because of unsupported optional features (10)
[   86.380089] BTRFS error (device sdb2): open_ctree failed
```

mkreisl, 2017-07-25:
I suppose you want to mount XBian sd-card on a separate computer, right?

XBian is using LZ4 as compression algorythm, which is not supported by standard linux kernel(s). That's the reason why you can not mount the volume.

You have to use a special kernel with BTRFS LZ4 patch for mounting the card, patches can be found here: https://github.com/xbianonpi/xbian-patches/tree/master/kernel

vondrt4, 2017-07-25:
If you don't want to compile a kernel from source, you can use another Xbian to mount the SD card created by Xbian.

To be honest, I have changed back to LZO in my installation because it's a real PITA when you need to rescue the system. Sorry, Manfred.

mkreisl, 2017-07-25:
> If you don't want to compile a kernel from source, you can use another Xbian to mount the SD card created by Xbian.

Of course, you can do this 😄 

> To be honest, I have changed back to LZO in my installation because it's a real PITA when you need to rescue the system. Sorry, Manfred.

There is no need to apologize, I did the same long time ago for my 'production' systems. Only test and IMX6 systems running with LZ4 compression. And of course, my homeserver is able to mount BTRFS LZ4 volumes.

But you have to switch back to LZO compression after flashing card BEFORE booting XBian the first time !

aladar42, 2017-07-26:
> I suppose you want to mount XBian sd-card on a separate computer, right?

I'm just trying to put the SD card into my PC so I can copy my movies over to the SD card storage. Is that not supported?

mkreisl, 2017-07-26:
Please read my first post here.

You can access sd-card very easily by predefined samba shares

aladar42, 2017-07-26:
So if I understand it correctly, my only two options are to patch my kernel, or set up samba, which will probably limit the data transfer significantly? Could I just ask why not have the SD card function  normally as a mass storage device as other distros do? This seems... Suboptimal.

mkreisl, 2017-07-28:
> So if I understand it correctly, my only two options are to patch my kernel, or set up samba, which will probably limit the data transfer significantly? Could I just ask why not have the SD card function normally as a mass storage device as other distros do? This seems... Suboptimal.

XBian already comes with Samba configured

There was a decision long time ago for using LZ4 as compression, don't ask me why. Of course, you can use LZO as well, it's just a configuration parameter as already told

Suboptimal? IMO it is suboptimal, to shut down RPi, pull out the sd-card, go to my PC, plug in the card into sd-card reader and mount device, copy files to sd-card, sync card, umount card, move card again to RPi and boot RPi again - that is pre-historical behavior for people never heard anything about lan connections ... - sorry @aladar42

closed 2017-07-28 by mkreisl