xbianonpi/xbian #847: deluged crashes

delugeduser, 2017-07-27:
Has basically been unusable for 3  or 4 months. Crashes with log:

[INFO    ] 14:15:00 configmanager:70 Setting config directory to: /var/lib/deluged/config
[ERROR   ] 14:15:04 main:237 importing '/usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabihf.so': /usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabihf.so: symbol SSLv2_client_method, version OPENSSL_1.0.0 not defined in file libssl.so.1.0.0 with link time reference
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/deluge/main.py", line 229, in start_daemon
    from deluge.core.daemon import Daemon
  File "/usr/lib/python2.7/dist-packages/deluge/core/daemon.py", line 39, in <module>
    from twisted.internet import reactor
  File "/usr/lib/python2.7/dist-packages/twisted/internet/reactor.py", line 38, in <module>
    from twisted.internet import default
  File "/usr/lib/python2.7/dist-packages/twisted/internet/default.py", line 56, in <module>
    install = _getInstallFunction(platform)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/default.py", line 44, in _getInstallFunction
    from twisted.internet.epollreactor import install
  File "/usr/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 24, in <module>
    from twisted.internet import posixbase
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 24, in <module>
    from twisted.internet import error, udp, tcp
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 29, in <module>
    from twisted.internet._newtls import (
  File "/usr/lib/python2.7/dist-packages/twisted/internet/_newtls.py", line 21, in <module>
    from twisted.protocols.tls import TLSMemoryBIOFactory, TLSMemoryBIOProtocol
  File "/usr/lib/python2.7/dist-packages/twisted/protocols/tls.py", line 41, in <module>
    from OpenSSL.SSL import Error, ZeroReturnError, WantReadError
  File "/usr/lib/python2.7/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import rand, crypto, SSL
  File "/usr/lib/python2.7/dist-packages/OpenSSL/rand.py", line 11, in <module>
    from OpenSSL._util import (
  File "/usr/lib/python2.7/dist-packages/OpenSSL/_util.py", line 4, in <module>
    binding = Binding()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 89, in __init__
    self._ensure_ffi_initialized()
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py", line 113, in _ensure_ffi_initialized
    libraries=libraries,
  File "/usr/lib/python2.7/dist-packages/cryptography/hazmat/bindings/utils.py", line 80, in build_ffi
    extra_link_args=extra_link_args,
  File "/usr/lib/python2.7/dist-packages/cffi/api.py", line 340, in verify
    lib = self.verifier.load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 75, in load_library
    return self._load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 151, in _load_library
    return self._vengine.load_library()
  File "/usr/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 149, in load_library
    raise ffiplatform.VerificationError(error)
VerificationError: importing '/usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabihf.so': /usr/lib/python2.7/dist-packages/cryptography/_Cryptography_cffi_813c10e0x7adb75f8.arm-linux-gnueabihf.so: symbol SSLv2_client_method, version OPENSSL_1.0.0 not defined in file libssl.so.1.0.0 with link time reference

delugeduser, 2017-07-27:
deluged: 1.3.10
libtorrent: 0.16.18.0

mkreisl, 2017-07-27:
@delugeduser 
What kind of hardware do you have? I suppose Cubox-i/Humminboard (i.MX6)

delugeduser, 2017-07-28:
Yes, Cubox i4

mkreisl, 2017-07-28:
There is a pin setting in file `/etc/apt/preferences.d/xbian` (Please don't ask why its there)

Please modify this file to something like this

```
....
Package: xopenssl xlibssl1.0.0 irqbalance-imx
Pin: release o=XBian
Pin-Priority: 1001
```

add `x` to openssl and libssl1.0.0 

and run `sudo apt-get update; sudo apt-get upgrade` again

delugeduser, 2017-07-29:
**apt-get upgrade**
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Worked like a charm. Very obscure. Thank you very much for solving this conundrum.

closed 2017-07-29 by delugeduser

delugeduser, 2018-03-06:
As a postscript, I need to add that xbian-package-repo resets the /etc/apt/preferences.d/xbian file, so the edit is required again.

mkreisl, 2018-03-06:
> As a postscript, I need to add that xbian-package-repo resets the /etc/apt/preferences.d/xbian file, so the edit is required again.

That's right, you have to use your own file to overwrite settings in xbian , in forum are some samples how to do that

For example :
```
cat /etc/apt/preferences.d/zz-user
Package: openssl libssl1.0.0
Pin: release o=XBian
Pin-Priority: -1

Package: openssl libssl1.0.0
Pin: release o=Debian
Pin-Priority: 1001
```