xbianonpi/xbian #848: NOOBS Download is not working

florian-asche, 2017-07-28:
Hello,

i did try to install the latest version for my rpi3, but the download failed.

mkreisl, 2017-07-28:
Was just downloading `XBian_by_NOOBS_v2.4_rpi3` with Linux XBian installer, absolutely no problem

florian-asche, 2017-07-28:
I mean after starting NOOBS and select xbian for installation. That download failed.

mkreisl, 2017-07-28:
Hmmm, seems you have serious network issues.

Downloaded and flashed NOOBS - runs perfect
Booting NOOBS, downloaded and installed XBian - went absolutely smooth
Starting XBian - works like a charm

florian-asche, 2017-07-29:
very curious!
ping is ok, speedtest is ok.

florian-asche, 2017-07-29:
Thanks for all the help. i will close this issue. In case someone else some years later read this, and did have the same issue, your device is maybe broken.

closed 2017-07-29 by florian-asche