xbianonpi/xbian #849: apt issue

florian-asche, 2017-07-28:
Hello,

After problems with NOOBS i did download the image without and installed it.

After booting up, i did try to make a "apt-get update" and it failed:
```
root@wohnzimmer ~ # apt-get update
Ign:1 http://deb.debian.org/debian stretch InRelease
Holen:2 http://deb.debian.org/debian stretch Release [118 kB]
Holen:3 http://deb.debian.org/debian stretch Release.gpg [2.373 B]
OK:4 http://apt.xbian.org stable InRelease               
Ign:3 http://deb.debian.org/debian stretch Release.gpg
Holen:5 http://deb.debian.org/debian stretch/main armhf Packages [9.272 kB]
Fehl:5 http://deb.debian.org/debian stretch/main armhf Packages 
  Hash-Summe stimmt nicht überein
  Hashes of expected file:
   - Filesize:9272411 [weak]
   - SHA256:500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d
   - MD5Sum:dd4fefed9163e5e515315f96a8bca783 [weak]
  Hashes of received file:
   - SHA256:68cd035eb68c60247e9999a2952156310f00e16b5ec81ed8ef9afd08521ff990
   - MD5Sum:70f67e9bd78d1484d3968d5ca196291b [weak]
   - Filesize:9272411 [weak]
  Last modification reported: Sat, 22 Jul 2017 08:08:59 +0000
  Release file created at: Sat, 22 Jul 2017 09:02:12 +0000
Holen:6 http://deb.debian.org/debian stretch/main Translation-de [1.620 kB]
Holen:6 http://deb.debian.org/debian stretch/main Translation-de [1.620 kB]
Holen:6 http://deb.debian.org/debian stretch/main Translation-de [1.620 kB]
Holen:6 http://deb.debian.org/debian stretch/main Translation-de [1.620 kB]
Fehl:6 http://deb.debian.org/debian stretch/main Translation-de
  
Holen:10 http://deb.debian.org/debian stretch/main Translation-en [5.393 kB]
Fehl:10 http://deb.debian.org/debian stretch/main Translation-en
  
Es wurden 16,4 MB in 4 s geholt (3.285 kB/s).
Paketlisten werden gelesen... Fertig
W: GPG-Fehler: http://deb.debian.org/debian stretch Release: Die folgenden Signaturen waren ungültig: BADSIG 8B48AD6246925553 Debian Archive Automatic Signing Key (7.0/wheezy) <ftpmaster@debian.org>
W: The repository 'http://deb.debian.org/debian stretch Release' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Fehlschlag beim Holen von http://deb.debian.org/debian/dists/stretch/main/binary-armhf/by-hash/SHA256/500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d  Hash-Summe stimmt nicht überein
   Hashes of expected file:
    - Filesize:9272411 [weak]
    - SHA256:500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d
    - MD5Sum:dd4fefed9163e5e515315f96a8bca783 [weak]
   Hashes of received file:
    - SHA256:68cd035eb68c60247e9999a2952156310f00e16b5ec81ed8ef9afd08521ff990
    - MD5Sum:70f67e9bd78d1484d3968d5ca196291b [weak]
    - Filesize:9272411 [weak]
   Last modification reported: Sat, 22 Jul 2017 08:08:59 +0000
   Release file created at: Sat, 22 Jul 2017 09:02:12 +0000
E: Fehlschlag beim Holen von http://deb.debian.org/debian/dists/stretch/main/i18n/by-hash/SHA256/cf2ee8cac943a8cd35b99526aae447acd85f55a1fc04325c0d2d241a26e6f698  
E: Fehlschlag beim Holen von http://deb.debian.org/debian/dists/stretch/main/i18n/by-hash/SHA256/6c100b8cb0d198295f3f0b632451f85feaf80a908e21006d7bc1fbaaa2135d3f  
E: Einige Indexdateien konnten nicht heruntergeladen werden. Sie wurden ignoriert oder alte an ihrer Stelle benutzt.
```

mkreisl, 2017-07-28:
Everything is working ok here, seems your provider or name server has some issues with resolution of deb.debian.org

```
LC_ALL=C apt-get update
Hit:1 http://apt.xbian.org stable InRelease
Hit:3 http://apt.xbian.org staging InRelease
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch Release
Reading package lists... Done
```

florian-asche, 2017-07-28:
no success too.

```
root@wohnzimmer /etc/apt # apt-get update
OK:1 http://apt.xbian.org stable InRelease
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease                 
Holen:3 http://cdn-fastly.deb.debian.org/debian stretch Release [118 kB]
Holen:4 http://cdn-fastly.deb.debian.org/debian stretch Release.gpg [2.373 B]
Holen:5 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Packages [9.272 kB]
Fehl:5 http://cdn-fastly.deb.debian.org/debian stretch/main armhf Packages
  Hash-Summe stimmt nicht überein
  Hashes of expected file:
   - Filesize:9272411 [weak]
   - SHA256:500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d
   - MD5Sum:dd4fefed9163e5e515315f96a8bca783 [weak]
  Hashes of received file:
   - SHA256:8539d0942b2db1431c39427d9e7f27b3872a82f8d432260b7ec4bf09048ffda1
   - MD5Sum:2b5da7703e8507af514afaf1a988df4f [weak]
   - Filesize:9272411 [weak]
  Last modification reported: Sat, 22 Jul 2017 08:08:59 +0000
  Release file created at: Sat, 22 Jul 2017 09:02:12 +0000
Holen:6 http://cdn-fastly.deb.debian.org/debian stretch/main Translation-en [5.393 kB]
Fehl:6 http://cdn-fastly.deb.debian.org/debian stretch/main Translation-en
  
Holen:7 http://cdn-fastly.deb.debian.org/debian stretch/main Translation-de [1.620 kB]
Fehl:7 http://cdn-fastly.deb.debian.org/debian stretch/main Translation-de
  
Holen:8 http://cdn-fastly.deb.debian.org/debian stretch/main Translation-de_DE [830 B]
Holen:9 http://cdn-fastly.deb.debian.org/debian stretch/non-free armhf Packages [70,4 kB]
Holen:10 http://cdn-fastly.deb.debian.org/debian stretch/non-free Translation-en [79,2 kB]
Es wurden 16,6 MB in 5 s geholt (2.919 kB/s).
Paketlisten werden gelesen... Fertig
E: Fehlschlag beim Holen von http://cdn-fastly.deb.debian.org/debian/dists/stretch/main/binary-armhf/by-hash/SHA256/500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d  Hash-Summe stimmt nicht überein
   Hashes of expected file:
    - Filesize:9272411 [weak]
    - SHA256:500b1d9e94593bd5dee7248c8a90a5e4c197f303c16a11719283d613b829d21d
    - MD5Sum:dd4fefed9163e5e515315f96a8bca783 [weak]
   Hashes of received file:
    - SHA256:8539d0942b2db1431c39427d9e7f27b3872a82f8d432260b7ec4bf09048ffda1
    - MD5Sum:2b5da7703e8507af514afaf1a988df4f [weak]
    - Filesize:9272411 [weak]
   Last modification reported: Sat, 22 Jul 2017 08:08:59 +0000
   Release file created at: Sat, 22 Jul 2017 09:02:12 +0000
E: Fehlschlag beim Holen von http://cdn-fastly.deb.debian.org/debian/dists/stretch/main/i18n/by-hash/SHA256/6c100b8cb0d198295f3f0b632451f85feaf80a908e21006d7bc1fbaaa2135d3f  
E: Fehlschlag beim Holen von http://cdn-fastly.deb.debian.org/debian/dists/stretch/main/i18n/by-hash/SHA256/cf2ee8cac943a8cd35b99526aae447acd85f55a1fc04325c0d2d241a26e6f698  
E: Einige Indexdateien konnten nicht heruntergeladen werden. Sie wurden ignoriert oder alte an ihrer Stelle benutzt.
```

```
root@wohnzimmer /etc/apt # cat sources.list
deb http://http.debian.net/debian stretch main non-free
```

mkreisl, 2017-07-28:
Hmmm, seems that the CDN does not work. You could try a different debian mirror, for example http://ftp.de.debian.org/debian/

florian-asche, 2017-07-29:
Hi, i am now quite sure, that this is linked to https://github.com/xbianonpi/xbian/issues/848.

I did the same sd card, from my pi3 into the pi2 and apt-get update works fine now. Some strange playback issue (image crapy) are also gone.

florian-asche, 2017-07-29:
1) scp problem: Write failed: Broken pipe, lost connection
2) NOOBS download failed
3) apt-get update / downloading packages failed
4) video playback from samba server is running, but video is with bocks

florian-asche, 2017-07-29:
is there anyone running a pi3 with the latest image, to make sure, its not software, then i have a faulty pi3 hardware.

mkreisl, 2017-07-29:
Again, strange network issues

Are you using LAN or WLAN?

florian-asche, 2017-07-29:
I am using LAN.

mkreisl, 2017-07-29:
Already changed eth-cable, port of switch, power supply?

mkreisl, 2017-07-29:
> is there anyone running a pi3 with the latest image, to make sure, its not software, then i have a faulty pi3 hardware.

As already told, NOOBS, XBian install, short XBian test - everything ok with RPi2 and RPi3

florian-asche, 2017-07-29:
Thanks for testing and answering!

Ok... i did some tests:
- Replaced Network cable
- Replaced SD-Card
- Replaced Power Supply
- Checked core temperature with by fingers

all with no success to resolve the issue.

i whould say, that the lan interface is damaged.
[image]

mkreisl, 2017-07-29:
Agree, your RPi3 is broken

florian-asche, 2017-07-29:
The strange thing is, ping from the device to some other device, and ping from pc to the device, i see no errors. and the device did not report error packages if i look at ifconfig.

florian-asche, 2017-07-29:
Thanks for all the help. i will close this issue.

closed 2017-07-29 by florian-asche