xbianonpi/xbian #851: Upgrade to stretch results in failing nfs access

ssakatausa, 2017-09-06:
Yesterday, I upgraded the RPi3 Xbian system from jessie to stretch and then found that access to NFS shares no longer worked. Checking the log file, I learned that kodi failed to load libnfs8 when trying to access the NFS shares. Once I installed libnfs8, the problem was gone. I suspect that dependency of xbian-package-xbmc on libnfs4 was the cause of the problem. If it is so, please make appropriate corrections.

Thanks,

  Shinichi

mkreisl, 2017-09-06:
@ssakatausa 

Yes,  upgrade process randomly does not upgrade libnfs4 to libnfs8, I do not know the reason for this behavior. 

I have 4 systems upgraded multtiple times, and on three of then libnfs8 was there, but on the 4th one libnfs4 was still there (I noticed this some days ago)

Currently there is dependency **libnfs4 | libnfs8**  in package xbian-package-xbmc, so a dist-upgrade (that's what the xbian-config GUI does when upgrading Jessie -> Stretch), the libnfs package should always be upgraded

How did you upgrade your system?

ssakatausa, 2017-09-06:
@mkreisl 

Initially, I used the upgrade option in Xbian's configuration menu. Because it resulted in a system with broken NFS access, I went back to jessie using brtfs's roll back facility. I then tried "apt-get update & apt-get upgrade & apt-get dist-upgrade" through ssh access to the Rpi3. In both cases, libnfs was not upgraded.

If the dependency is on libnfs4|libnfs8, apt-get would not touch libnfs in my understanding, as long as the latest version of either libnfs4 or libnfs8 is installed. At the same time, the log file I saw indicates that the current version of kodi does not use libnfs4 at all. If this is the case, shouldn't the dependency on libnfs8 be eliminated?

mkreisl, 2017-09-07:
@ssakatausa Thanks for your answer

> system indicates that the current version of kodi wants libnfs8 and does not use libnfs4 at all. If this is the case, shouldn't the dependency on libnfs4 be eliminated?

In the future - yes. But currently, xbian-package-xbmc is still build for Stretch and Jessie, so this was the best option doing this. 

I will think about that, maybe I'll find a better solution

mkreisl, 2017-09-08:
I solved this yesterday by modifying those dependencies during build time

closed 2017-09-08 by mkreisl