xbianonpi/xbian #852: [cubox-i] audio drops every 15 sec. after kernel update

didlich, 2017-10-15:
Today I updated to linux-image-armmp version 4.8.15+rt19+-1498667745.
Now there is an audio drop every 15 seconds in online streams.

A similar problem was already reported in #833 , which was closed.

What could be the problem?

mkreisl, 2017-10-15:
@didlich 

You updated kernel package only? I can't believe that

Which Debian are you running - Jessie or Stretch

Which Kodi version are you running?

didlich, 2017-10-15:
```bash
lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.9 (jessie)
Release:        8.9
Codename:       jessie

cat /etc/*-release
PRETTY_NAME="XBian 1.0 (knockout)"
NAME=XBian
VERSION_ID=1.0
VERSION="1.0 (knockout)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.xbian.org/"
SUPPORT_URL="http://forum.xbian.org/"
BUG_REPORT_URL="https://github.com/xbianonpi/xbian/issues"

```

Kodi version is: 17.4 Git-2017-08-29

mkreisl, 2017-10-15:
Ah, ok, probably it's the Kodi version. Some users already reported similar issues with this version (unfortunately I'm not able to reproduce it)

So, you're still running Debian Jessie, the only way for you is downgrading Kodi to version 17.3

```
sudo apt-get install xbian-package-xbmc=17.3-1496273433
```

I've built new test version, but this version is available for Debian Stretch only

didlich, 2017-10-15:
OK, thanks for your help. I'll try the version of Kodi you mentioned.

Would you suggest to upgrade to Debian Stretch? How can I do this in xbian in a smooth way?

mkreisl, 2017-10-15:
> Would you suggest to upgrade to Debian Stretch? How can I do this in xbian in a smooth way?

Sure, end of this year no new XBian packages are build for Jessie.

You should get a popup window (that requires all packages have to be uptodate)

See here www.xbian.org/73-2/

didlich, 2017-10-15:
Thank you to point me in the right direction.

mkreisl, 2017-10-15:
After you have upgraded to Stretch, you could install this test version http://apt.xbian.org/pool/devel/imx6-stretch/x/xbian-package-xbmc/xbian-package-xbmc_17.4-1508082761_armhf.deb I already mentioned to see if your issue has been gon

Important:
Before installing this version, you have downgrade all cec packages to version 4.0.2.1~xbian-1489608861, or you can run that script:

```
#!/bin/bash

KV=xbian-package-xbmc_17.4-1508082761_armhf.deb
CV=4.0.2.1~xbian-1489608861

stop xbmc
apt-get update
apt-get install --yes --allow-downgrades --reinstall cec-utils=$CV libcec=$CV libcec-dev=$CV libcec4=$CV python-libcec=$CV
wget http://apt.xbian.org/pool/devel/imx6-stretch/x/xbian-package-xbmc/$KV -O /tmp/xbian-package-xbmc.deb
dpkg -i /tmp/xbian-package-xbmc.deb
rm -f /tmp/xbian-package-xbmc.deb
start xbmc
```

didlich, 2017-10-17:
@mkreisl, decided to use a fresh install of xbian from current image for cubox, but the problem remains

mkreisl, 2017-10-17:
@didlich 

Are you using my test version or just the version shipped by the image?

didlich, 2017-10-18:
I've started with the default version shipped by the image.
There is also a problem with video in the same moment.

The next step is to install your test version. I'll report the result.

didlich, 2017-10-18:
@mkreisl , tested your version :-)
great, works well, problem is gone

mkreisl, 2017-10-18:
@didlich 

Very good news, thanks for testing 😄

closed 2017-10-18 by didlich