xbianonpi/xbian #853: After linux-image-bcm2836_4.9.58+ upgrade system not booting

er0l, 2017-10-23:
Latest linux-image-bcm2836_4.9.58+ after installation will not boot anymore, screen blank, no activity.

None of the rescue modes work (rescue,rescue_early,rescue_late), neither holding shift while booting.

Issue is definitely with the kernel since I was able to restore it by copying fresh xbianboot partition from  XBian_Latest_rpi3.img.gz where 4.9.56+ is used.

After it booted and network is up then I downgraded both kernel and headers from rpi2-stretch repository by hand.

This needs to be looked at since I could reproduce the issue couple of times by upgrading again to 4.9.58+.

Bastlwastl2010, 2017-10-23:
Same issue here on a rpi 2

mkreisl, 2017-10-23:
Ok, seems that this is a broken kernel, will have to check this

er0l, 2017-10-23:
Initramfs is enabled in config.txt

mkreisl, 2017-10-23:
Just tested kernel. Have to confirm that kernel 4.9.58 (build?) is definitely broken. Checked build logs, everything looks ok. Have to investigate what's going wrong there. Broken kernel deleted from Repo

er0l, 2017-10-23:
Thanks for your help and hard work. Keep it up.

mkreisl, 2017-10-23:
Found culprit, see https://github.com/raspberrypi/linux/issues/2239

er0l, 2017-10-24:
I have just upgraded to 4.9.58+-1508858583 and can confirm the issue is fixed now.

vin-c, 2017-10-28:
Same issue with rpi2 and rpi3.
Copied `kernel.img` & `initrd.gz` from last downloadable image to root partition, booting fine.
Then update to last 4.9.58+-1508858583 and reboot.
Everything seems ok now !
Thank you

closed 2017-10-30 by mkreisl