xbianonpi/xbian #854: xbian restarts when scrollig movies screen

mikemorrell2, 2017-12-03:
I am using Xbain Jessie version 20171202-0 on iCubox (iMX6). This is a fresh install with no plugins. I setup my movie library and let it sync with IMDB. After that was done, I went to the Movies main screen and scrolled down to the end using keyboard. After navigating and scrolling around the menus with keyboard input for awhile, the whole system reboots. This is 100% repeatable. I do not know what logs would help to analyze the problem.

mkreisl, 2017-12-03:
@mikemorrell2 
You installed an outdated Jessie image, why did you do that?

mikemorrell2, 2017-12-04:
The link on the download page for the Cubox-i board gave me that version. Is there a newer version I should be using?

mkreisl, 2017-12-04:
You're confusing me. On download page you get XBian_Latest_imx6.img.gz, which was built Dec 1 and contains Stretch, not Jessie.

So, please make sure that you install correct version before continuing here. Thanks

mikemorrell2, 2017-12-04:
cat /etc/xbian-codename gives jessie as output.
On the Kodi system info screen I see Operating System XBain 1.0(knockout) kernel linux 4.8.15+rt9+.
Build is Kodi 17.6 Git:2017-11-18-433b93a compiles Nov. 19, 2017.

mkreisl, 2017-12-06:
> cat /etc/xbian-codename gives jessie as output.
> On the Kodi system info screen I see Operating System XBian 1.0(knockout) kernel linux 4.8.15+rt9+.
> Build is Kodi 17.6 Git:2017-11-18-433b93a compiled Nov. 19, 2017.

Oops, you're right. There was an outstanding PR, but forgot this to fix this issue till today. But OC, this is only cosmetic

But anyway, I can not reproduce your issue, can you tell me a bit more about your system?

Where are the vids located,
Are you using mysql database,
What about CPU/RAM (cores?)

On my system, a Hummingboard pro, 2GB RAM quad core, fs on SATA disk,  everything works well

mkreisl, 2017-12-07:
Hmmm, that's weird

Ahh, so CPU and RAM is identical to mine