xbianonpi/xbian #856: Touchscreen calibration

joeyb197, 2018-01-15:


joeyb197, 2018-01-15:
[image]

mkreisl, 2018-01-15:
@joeyb197 That's working for Kodi v17
http://lektiondestages.blogspot.de/2016/06/properly-setting-up-5-hdmi-touchscreen.html

joeyb197, 2018-01-15:
I'm new in these things, so how can I open the screen where I put commands?

joeyb197, 2018-01-15:
Oh i'ts not command, how can I find these files then? :/

mkreisl, 2018-01-15:
and this is good thread also: http://forum.xbian.org/thread-3849-post-33175.html

mkreisl, 2018-01-15:
/home/xbian/.kodi/userdata

joeyb197, 2018-01-15:
Yeah, but where do I access these files? how?

mkreisl, 2018-01-15:
OMG

1) close Kodi and login (user xbian) -- you need keyboard, or
2) ssh into XBian (you need LAN connection)

editor you can use nano

mkreisl, 2018-01-15:
Unfortunately my touch has bee broken several months ago, so I can not give more help

closed 2018-12-04 by mkreisl