xbianonpi/xbian #857: How to build the custom Xbian Image

amanharitsh123, 2018-03-05:
I have made some changes to the kodi fork https://github.com/amanharitsh123/xbmc  . I want to make a custom xbian image using this fork and want to test my changes. How can i do this ?
Please Help

mkreisl, 2018-03-05:
@amanharitsh123 

Do you have already built Kodi from your source? If you did it, you do not need to build an XBian image, it's just enough to install your custum kodi deb package (xbian-package-xbmc) via dpkg -i

If you still have to build your own kodi package, please read an follow these instructions: https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages

amanharitsh123, 2018-03-05:
@mkreisl 
Thanks for the reply 
I want to build kodi isengard from source https://github.com/amanharitsh123/xbmc
How can i make deb package using my repo(https://github.com/amanharitsh123/xbmc)
Can you please help?

amanharitsh123, 2018-03-05:
and after building a deb package , can i directly install that on my rpi running xbian ?
Should i have to remove the kodi that  my xbian image already had ?
@mkreis

mkreisl, 2018-03-05:
Ah, ok.

Step 1:
Please read an follow these instructions: https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages. This installs XBian development environment

Step 2:
You have to clone this repository: https://github.com/xbianonpi/xbian-package-xbmc
This is the repository to build Kodi

Step 3:
You need our `treeIsengard` tree from here https://github.com/xbianonpi/xbian-sources-xbmc, because there are many patches to make it work under XBian. So you have to fork it and make your own changes in `treeIsengard' branch

Step 4
Assuming you want to build package for Pi 2/3, you have to modify the `config_source_addr` setting in this file https://github.com/xbianonpi/xbian-package-xbmc/blob/master/build/rpi2-cross-isengard/config (has to point to your forked and modified repo)

Step 5:
To build package, you have to run $xbiangit --arch rpi2-cross-isengard

Please note that Kodi Isengard is very very out of date, so I can not guarantee that building of this version still works

mkreisl, 2018-03-05:
> and after building a deb package , can i directly install that on my rpi running xbian ?
Yes

> Should i have to remove the kodi that my xbian image already had ?
No

amanharitsh123, 2018-03-05:
@mkreisl Thanks a lot 
Its really a great help to me

amanharitsh123, 2018-03-05:
With the development setup guide , https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages
It doesnt found any command "xbiangit"

mkreisl, 2018-03-05:
You have to log out and log in again

The command is `$xbiangit`

amanharitsh123, 2018-03-05:
Tried that , looks like i have to add the path to $PATH

mkreisl, 2018-03-05:
$xbiangit is not a command, it's a environment variable and points to run.me

amanharitsh123, 2018-03-05:
it means that echo $xbiangit must return some path then?

amanharitsh123, 2018-03-05:
i did a chmod 777 /etc/profile.d/xbiangit.sh , it solved the issue

mkreisl, 2018-03-05:
> it means that echo $xbiangit must return some path then?

Exactly

mkreisl, 2018-03-05:
> i did a chmod 777 /etc/profile.d/xbiangit.sh , it solved the issue

Hmm, thats weird:

```
manfred@kmnote5:/usr/src/xbian/xbian-package-kernel> ll /etc/profile.d/xbiangit.sh 
-rw-r--r-- 1 root root 273  1. Mai 2017  /etc/profile.d/xbiangit.sh
```

amanharitsh123, 2018-03-05:
No , there is something wrong . Let me check

amanharitsh123, 2018-03-05:
I have found the issue , we cant run $xbiangit with the root user. After loggin in again , we have to continue the build process through as a user , not root

mkreisl, 2018-03-05:
Ehm, working under root is usually never a good idea :smile:

amanharitsh123, 2018-03-05:
i think there is a typo here "cd ../xbian-prep-root; $xbiangit --arch [platform] --action build" , shouldnt it be the "cd . " rather than "cd .."

https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages
Step 4

mkreisl, 2018-03-05:
No, that command installs your schroot build environment. You need this for building binary packages.
Btw, XBian usually does not use a cross compiler (expect building kernel packages)

amanharitsh123, 2018-03-05:
and i there is this error now 
after running $xbiangit --arch [platform] --action build
/create.buildroot no such file or directory

mkreisl, 2018-03-05:
```
$xbiangit --arch amd64stretch --action build
```

amanharitsh123, 2018-03-05:
same error , I am using ubuntu 16.04

mkreisl, 2018-03-05:
Hmmm, I have run that command many many times, never had issues. I'm running Debian Stretch for doing that

amanharitsh123, 2018-03-05:
then let me try that on my debian vm

mkreisl, 2018-03-05:
But make sure that you are root when running that command

amanharitsh123, 2018-03-05:
"cd ../xbian-prep-root; $xbiangit --arch [platform] --action build
 in this command , we are in the this directory https://github.com/xbianonpi/xbian/tree/master/xbian-prep-root
 right ?

mkreisl, 2018-03-05:
we are in xbian-prep-root folder, right

amanharitsh123, 2018-03-05:
@mkreisl  the same error on debian jessie

mkreisl, 2018-03-05:
Hmmm, looks like your XBIANROOT variable has not been set properly

```
root@kmnote5-debian9:/usr/src/xbian/xbian-prep-root/extra-files# env | grep XBIAN
XBIANROOT=/usr/src/xbian
```

amanharitsh123, 2018-03-05:
should i clone this in /usr/src/

amanharitsh123, 2018-03-05:
XBIANROOT is set correctly , i have checked

mkreisl, 2018-03-05:
> should i clone this in /usr/src/

No, you can do this in your home directory

mkreisl, 2018-03-05:
Hmmm, at the moment I'm a bit confused. I think you're making an error but do not know what the error is

amanharitsh123, 2018-03-05:
Can you guess some possible error ?
Looks like the script is referring to a wrong directory for create.buildroot which is present in the parent of current working directory

mkreisl, 2018-03-05:
Haven't set up this enviromnent for a long time.

Please give me some time to install it on a fresh environment

mkreisl, 2018-03-05:
... and the build root procedure is broken at the moment due to some deb packages could not be found.

Have to fix this first

amanharitsh123, 2018-03-05:
oh alright , Thank you
please let me know if you find any fix :)

amanharitsh123, 2018-03-05:
Fix : sudo -E $xbiangit --arch [platform] --action build

mkreisl, 2018-03-05:
I have commited some fixes, https://github.com/xbianonpi/xbian/commit/80feea4a37416346c2b2300c4ac612a65bf58e90, https://github.com/xbianonpi/xbian/commit/4d3b20ff21771bdfe045fd76c2a340c8e289d9c7, https://github.com/xbianonpi/xbian-package-repo/commit/b2b64de561aeacae67078ffbd79e9cb4d47ce67a

and updated wiki article https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages

Now it should work without issues

Please note, the user has to be member of group `users`. If you don't want this, you have to edit
config files in `/etc/schroot/chroot.d` folder manually

amanharitsh123, 2018-03-06:
@mkreisl  Thanks a lot for the help :)

amanharitsh123, 2018-03-06:
I am gettting some error https://paste.fedoraproject.org/paste/8ir9TzMrLwM94OmgJEGo~A
after sudo -E $xbiangit --action build

mkreisl, 2018-03-06:
Hmmm, never get this issue, looks like that another process is accessing this folder. 

Probably not good idea to use `/home/osboxes/Desktop/XFS/xbian` as path, isn't Desktop your current desktop?

I think better path is /home/osboxes/Desktop

amanharitsh123, 2018-03-06:
and on ubuntu system is , 
Method gave invalid 103 redirect message after 20 mins

mkreisl, 2018-03-06:
Sorry, at this point I have no idea how to help.
I'm using openSUSE 13.1 (my local devel system), Debian Jessie (this is our main build system), Debian Stretch (VM for testing and ARM powered home server), never had this issue

Are any FS mounted by autofs?

amanharitsh123, 2018-03-06:
Probably not , I think this erro Method gave invalid 103 redirect messageis due to some broken ppa 
this happens while apt update checking for ftps

mkreisl, 2018-03-06:
Do you really run `sudo -E $xbiangit --action build`?
Why do you omit parameter --arch paramater

amanharitsh123, 2018-03-06:
Because i was not sure which one to choose for rpi3

mkreisl, 2018-03-06:
In your case **amd64stretch** is the best choice

I already mentioned this:

> Current Debian is Stretch, so you have to install amd64stretch root environment if you want to build Kodi package for Pi 2/3

amanharitsh123, 2018-03-06:
Well , i have something in my mind . 
i think emdebian.org doesnt contain armhf packages
for debian jessie 
we can add this as a mirror http://ftp.debian.org/debian/dists/jessie/main/binary-armhf/
But i dont know where to make changes

mkreisl, 2018-03-06:
emdebian.org supplies multiarch support (incl. cross compiler) for Debian Jessie, but this is not required for Debian Stretch, because Stretch comes with complete multiarch support already included.
And, you do not need to add anything as mirror

amanharitsh123, 2018-03-06:
@mkreisl  I badly need to compile  kodi-isengard which cant be done using stretch and debian chroot is ubale to apt-update from emdebian 
Is there anything i can do right now ?

mkreisl, 2018-03-06:
> I badly need to compile kodi-isengard which cant be done using stretch

Why - error messages?

Did you have sucessfully set up root environment **amd64stretch**?

And of course, you have to edit file `.../xbian-package/xbmc/build/rpi2-cross-isengard/config`, like this

```
...
config_build_env=schroot -c amd64stretch -p --
config_source_branch=treeIsengard
config_source_addr=https://github.com/xbianonpi/xbian-sources-xbmc.git
...

amanharitsh123, 2018-03-06:
I have done that , now there is a error for 
no rule to make target 'java-not-found ', needed by 'codegenerator'

mkreisl, 2018-03-06:
Hmmm, so seems that Isengard can not be build for Debian Stretch. I was trying it also, and my build stopped during configure process (havent looked deeper, does not make sense for me).

So, I'm currently building Isengard for Debian Jessie, that seems to work. But, unfortunately setting up root (amd64jessie) fails (got same Method gave invalid 103 redirect message errors), but I have already fixed that

You can apply the fix and locally try to setup amd64jessie again
```
manfred@kmnote5-debian9:/usr/src/xbian$ git diff create.buildroot
diff --git a/create.buildroot b/create.buildroot
index ea62cbb..42abad8 100644
--- a/create.buildroot
+++ b/create.buildroot
@@ -165,7 +165,7 @@ do_other()
     esac
     LC_ALL=C schroot --directory / -p -c $config_platform -- apt-get update 2>/dev/null || ( \
         case "$config_platform" in
-            *stretch*)
+            *stretch*|*jessie*)
                 # As it was written (for Debian Stretch) apt-get does not work with method mirror://
                 schroot --directory / -c $config_platform -p -- sed -i -e 's%mirror://apt.xbian.org/mirror.txt%http://apt.xbian.org/%g' /etc/apt/sources.list.d/xbian.list
  ```

Seems to work, but Isengard is not compatible with libcec4 library. Before building, libcec4 has to be removed and libcec3 installed manually in root amd64jessie

amanharitsh123, 2018-03-07:
Which Linux distro are you using ? 
I am still getting this error Method gave invalid 103 redirect message  after 
sudo -E $xbiangit --arch amd64jessie --action build

mkreisl, 2018-03-07:
Debian 9 VM for testing, fresh setup of amd64jessie root.

You have to remove amd64jessie folder (sudo -E $XBIANROOT/build-root-dirs/amd64jessie) and setup root amd64jessie again. Did you do that?

amanharitsh123, 2018-03-07:
Yes , I did that . but still the error with 
GET ftp.de.debian.org jessie/main/amd64 packages

amanharitsh123, 2018-03-07:
Thw same error if i try to apt-get update in schroot

mkreisl, 2018-03-07:
Please show me file /etc/apt/sources.list.d/xbian.list

amanharitsh123, 2018-03-07:
https://paste.fedoraproject.org/paste/OVq7hdszRR7uei2EQ-HoFw

[62 more replies]

closed 2018-03-11 by amanharitsh123