xbianonpi/xbian #858: [imx6] ahci-imx not working?

chr-b, 2018-03-12:
I am connecting an external hard drive via eSATA to my cubox i2 - but the Xbian Kernel is not detecting anything.

dmesg does not provide any life sign, especially when connecting the drive to the cubox. There is absolutely no output from ahci-imx:
``
dmesg | grep ahci
[    0.000000] Kernel command line: console=ttymxc0,115200 console=tty1 coherent_pool=2M ahci_imx.hotplug=1 raid=noautodetect telnet root=/dev/mmcblk0p2 video=mxcfb0:dev=hdmi,1920x1080M@60,if=RGB24 rootwait consoleblank=0 selinux=0 quiet loglevel=0 nohdparm splash partswap dmfc=3 pcie_aspm=off net.ifnames=0 biosdevname=0 pci=nomsi mxc_hdmi.rgb_quant_range=full mxc_hdmi.enable_3d=enable vpu352=0 overclock=0 rootfstype=btrfs rootflags=subvol=root/@,autodefrag,compress=lz4 --startup-event mountall
``

The module seems to be already loaded though:
``
sudo modprobe ahci-imx --first-time
modprobe: ERROR: could not insert 'ahci_imx': Module already in kernel
``

I am using a recent XBian imx6 release:
``
$ uname -a
Linux xbian 4.8.15+rt9+ #1 SMP PREEMPT RT Wed Jun 28 18:30:56 CEST 2017 armv7l GNU/Linux
``

PS:
Connecting the same hard disk via USB to my cubox works fine.
Connecting the same hard disk via eSATA to my laptop works fine.
I have an external power cable for the hard disk itself.

mkreisl, 2018-03-12:
Already checked cable etc?

I do not have Cubox-i, but my Hummingboard works very well with my connected SATA disk for years.

chr-b, 2018-03-12:
Yes, sure, hard disk works fine when attached to Laptop using the same cable.
What I find curious is that the ahci-imx module does not show any output in dmesg. 
Or does this module only become active when a cable is actually connected to the eSATA port?

mkreisl, 2018-03-12:
Did you already tried rebooting Cubox while external disk is connected to see if that makes any difference?

If still no response, I suppose that your eSATA port of your Cubox-i is broken.

chr-b, 2018-03-12:
Can you do me favor and boot your device with disconnected eSATA cable?
I would just like to know if ahci-imx module still prints any output in dmesg or not.

mkreisl, 2018-03-13:
> Can you do me favor and boot your device with disconnected eSATA cable?

That's a bit complicated here, because my root fs is on this sata drive, and drive is not an external one

Will have to think about that

delugeduser, 2018-05-22:
I can confirm that with the cable disconnected, I get in dmesg:

> [    0.000000] Kernel command line: console=ttymxc0,115200 console=tty1 coherent_pool=2M ahci_imx.hotplug=1 raid=noautodetect telnet root=/dev/mmcblk0p2 video=mxcfb0:dev=hdmi,1920x1080M@60,if=RGB24 rootwait consoleblank=0 selinux=0 quiet loglevel=0 nohdparm splash partswap dmfc=3 pcie_aspm=off net.ifnames=0 biosdevname=0 pci=nomsi mxc_hdmi.rgb_quant_range=full mxc_hdmi.enable_3d=enable vpu352=0 overclock=0 rootfstype=btrfs rootflags=subvol=root/@,autodefrag,compress=lz4 nohz=off --startup-event mountall
> [    1.341417] ahci-imx 2200000.sata: fsl,transmit-level-mV value 1104, using 00000044
> [    1.341429] ahci-imx 2200000.sata: fsl,transmit-boost-mdB value 0, using 00000000
> [    1.341438] ahci-imx 2200000.sata: fsl,transmit-atten-16ths value 9, using 00002000
> [    1.341447] ahci-imx 2200000.sata: fsl,receive-eq-mdB not specified, using 05000000
> [    1.344036] ahci-imx 2200000.sata: SSS flag set, parallel bus scan disabled
> [    1.344067] ahci-imx 2200000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
> [    1.344079] ahci-imx 2200000.sata: flags: ncq sntf stag pm led clo only pmp pio slum part ccc apst
> [    1.345996] scsi host0: ahci-imx
> 

I guess this suggests your eSata port is bad/damaged.