xbianonpi/xbian #859: Install Xbian on Sony Blu Ray BDP-S3100

Kosay, 2018-05-04:
Hi all!

I bought Sony Blu ray BDO-S3100. It has Opera TV store and I use it, but Opera software will discontinue its support in 28th June! I contacted with Sony and Opera but there is no solution. I think if I using XBian OS will solve the problem. I take this information about the device.
--------- Details----------
Processor Sony CXD90018R-DA
SK Hynix H27U2G8F2CTR-BC NAND Flash Memory - 256 MB
Samsung K4B1G1646G-BCH9 DDR3 SDRAM Memory - 128 MB
Samsung K4B2G1646E-BCKO DDR3 SDRAM Memory - 256 MB
MediaTek RT3370L Single-Chip WiFi 802.11b/g/n 1T1R w/ USB

http://techinsights.com/uploadedFiles/Public_Website/Content_-_Primary/Teardowncom/Sample_Reports/Digital_Home_Survey_Sample_Report.pdf

--------------------
Can I install Linux and Kodi on this device?

mkreisl, 2018-05-04:
No

closed 2018-05-04 by mkreisl

The-Exnor, 2018-05-04:
Just for curiosity what is inside of the Sony/MediaTek SoC ? Can't find any specs online :/ (although i can find retailers selling the chip)

mkreisl, 2018-05-04:
> Just for curiosity what is inside of the Sony/MediaTek SoC ? Can't find any specs online :/ (although i can find retailers selling the chip)
> @mkreisl

Silicon :smiley:

The-Exnor, 2018-05-04:
@mkreisl  lol sure, but i was wondering about specs :P

Kosay, 2018-05-09:
@The-Exnor . I do not know why Sony does not give any information about the device! Sony only gives the source code in its website (source code for libraries and kernel). I read the code for each spec, and I got the information by code. 
I think the best solution is getting new Apple TV :1st_place_medal: