xbianonpi/xbian #860: xbian Mirror 2 is a bust

delugeduser, 2018-05-21:
http://apt2.xbian.org/ xbian is either hacked or no longer a valid DNS resolve. Either way, this in sources.list will cause problems.

FYI for anyone getting something like this:

> 100% [Packages 33.2 MB] [1 InRelease gpgv 5961 B] [Waiting for headers]Splitting up /var/lib/apt/lists/partial/apt.xbian.org_mirroIgn mirror://apt.xbian.org stable InRelease
> E: GPG error: mirror://apt.xbian.org stable InRelease: Clearsigned file isn't valid, got 'NODATA' (does the network require authentication?)

Change /etc/apt/sources.list.d/xbian.list from

`deb mirror://apt.xbian.org/mirror.txt      stable  main    imx6-jessie    armv7l-jessie`

to

`deb http://apt.xbian.org/      stable  main    imx6-jessie    armv7l-jessie`

Should apt2.xbian.org be pulled from the mirrors?

f1vefour, 2018-05-25:
Thank you for the report, it is a known issue. We have lost contact with the mirror and have been deciding what to do, sorry for the inconvenience.

delugeduser, 2018-05-25:
No problems. I reported it more for all those who might encounter it and needed a work-around.

closed 2018-05-25 by delugeduser