xbianonpi/xbian #862: How to enable logging?

peci1, 2018-12-02:
I have a clean image of latest Xbian on RPi 1B :

```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.6 (stretch)
Release:        9.6
Codename:       stretch
```

    Linux xbian 4.14.80+ #1 PREEMPT Tue Nov 13 15:50:56 CET 2018 armv6l GNU/Linux

I see there's no system-wide logging service enabled, like syslog, rsyslog etc. I also can't find any logging service in xbian-config. journaltctl doesn't work because systemd doesn't have PID 1. Nor does the wiki help. 

What's currently the suggested way to enable system-wide logging (to get e.g. cron logs, network manager logs and so on)?

mkreisl, 2018-12-02:
@peci1 I'm using `rsyslog` for years (logging to common log server)

closed 2018-12-04 by mkreisl

peci1, 2018-12-04:
Thanks, that works. I just made sure logs are written to tmpfs...