xbianonpi/xbian #863: Security problem - sudo is needed to start kodi, why ?

plumpsack, 2018-12-04:
Hi,

why do I need "sudo" to start kodi ?

I renamed sudo from /usr/bin/sudo and figured out that xbmc(kodi will not start anymore in the gui.

Why?

mkreisl, 2018-12-04:
It's not a security problem

Kodi is running as user/group xbian, and this is done by sudo, [see here](https://github.com/xbianonpi/xbian-package-xbmc-scripts/blob/master/content/etc/init/xbmc.conf#L74)

If you set the variable DPRIORITY=yes in /etc/default/xbmc, sudo is not used

plumpsack, 2018-12-04:
ok, i set DPRIORITY=yes into /etc/default/xbmc
but after an reboot I get an xbmc-config-error message on my screen.
?

mkreisl, 2018-12-04:
Tell me, why do you rename sudo ???

That makes absolutely no sense, `xbian-config` for example (which is started per default if you log in as user xbian) is started via sudo, because root rights are required.

mkreisl, 2018-12-04:
> And that is exactly the problem.
Which  problem?

> 
> Why does a normal user needs "sudo" to run kodi?
That's the way it is

closed 2018-12-04 by mkreisl