xbianonpi/xbian #864: XBian 20190215-0 - CEC Keeps turning on TV

anselal, 2019-02-19:
Hi there,

I am running the latest XBian 20190215-0 and on this update, my TV keeps turning on although the CEC option to turn on the TV is off.

Is there perhaps an error with some library?

mkreisl, 2019-02-19:
@anselal There was an issue with the screensaver handling, you have to update **xbian-package-xbmc-scripts** to version 1.7.11

anselal, 2019-02-19:
`xbian-package-xbmc-scripts` is already in version 1.7.11. and there are not other updates available on my install.

Is there anything else I can do ? My temporary solution is to unplug the HDMI cable when I am not using xbian. Shutting it down is no option since I also run flexget on my RPi

anselal, 2019-02-20:
I disabled the screensaver and the problem was fixed but I think this is just a workaround.

mkreisl, 2019-02-20:
@anselal 
Please tell me more about your system
- Have you Kodi configured that screensaver should turnoff TV when activated? (in CEC settings)
- Is `SCREENOFF=yes` in file /etc/default/xbmc

anselal, 2019-02-20:
- Yes, I switch the TV manually off
- When the TV was turning on Kodi was active source
- This setting was turned on, I just turned it off
- There is no such option in my * /etc/default/xbmc
- My version is 18.1
- the log file had just this inside `xbmc-screensaver stop/pre-start, process 4914`

mkreisl, 2019-02-20:
Thanks for answering my questions.

So, external script `/etc/init/xbmc.screensaver.conf` is not involved (there was an issue in this script)
Does the TV still powering on after turning off the CEC setting?

anselal, 2019-02-20:
As mentioned earlier, disabling the screensaver did the job so I couldn't tell if turning off the CEC setting for the screensaver did something.

I guess we can close the issue.

Thnx for your time !!!

florian-asche, 2019-02-25:
Hi, 
just fyi, i have the same problem. Also i did disable the screensaver, that solved the problem.

mkreisl, 2019-02-25:
@florian-asche 
Please post your file `/home/xbian/.kodi/userdata/peripheral_data/rpi_2708_1001.xml`
and `/etc/default/xbmc`

You're doing the same. You turn off your TV manually  and after some time TV gets powered on, right?

florian-asche, 2019-02-26:
Hi, i did clean up the kodi directory some days ago. Restored from old system. Here are the results:

root@wohnzimmer:~# cat /home/xbian/.kodi/userdata/peripheral_data/rpi_2708_1001.xml
`cat: /home/xbian/.kodi/userdata/peripheral_data/rpi_2708_1001.xml: Datei oder Verzeichnis nicht gefunden`
root@wohnzimmer:~# cat /etc/default/xbmc
```
#
# This is controlfile for XBMC/Kodi application. Most variables
# can be set by Kodi's XBian config GUI. Please do not modify
# this file manually unless you know what you are doing.
#

DAEMON=$(test -x /usr/local/lib/kodi/kodi.bin && nm=kodi || nm=xbmc; echo /usr/local/lib/$nm/$nm.bin;)

# Userdefined script will be run before XBMC/Kodi is starting or has been stopped,
# or TV is turned on/off controlled by SCREENOFF variable. See template script
# /usr/local/share/doc/xbmc/user-run.sh
# Parameter 1: what, currently 'preload', 'preload' and 'screensaver'
#           2: action, currently 'start' and 'stop'
#
USERRUN="/usr/local/sbin/user-run.sh"

# Put name of all processes which XBMC/Kodi should wait for before it will be started.
# e.g. WAITFOR="mysqld process1 process2". Please note, if process configured here does
# not exist, start of Kodi will be delayed dramatically.
#
WAITFOR=""

# Niceness levels for XBMC/Kodi while playing and while in screensaver is active.
#
PPLAYER="-3"
PSAVER="7"

# Turn CEC capable TV off on XBMC/Kodi entering screensaver (after 'AFTER' minutes)
# If IGNOREPLAYER=yes, TV will be turned off even when media are currently played.
# Userdefined script (see USERRUN) will be run with screensaver start/stop.
#
SCREENOFF=no
IGNOREPLAYER=no
AFTER=3

# Stop XBMC/Kodi when the CEC capable TV is turned off. This will also start
# XBMC/Kodi again when it's turned on.
# It should be noted that this option will set 'standby_pc_on_tv_standby' to
# value 13009 in all peripheral_data/*.xml files of Kodi's profile
#
XBMCOFF=no

# Turn CEC capable TV on when XBMC/Kodi exits.
#
XBMCEXITON=no

# Turn CEC capable TV off on system shutdown.
# Please note, the SCROFFRUN parameter does not longer exist, if any executable
# script has been defined here, it has been moved into /lib/systemd/system-shutdown
# folder and is started on shutdown AND reboot event.
#
SCROFFHALT=no

# Enable dynamic priority handling of XBMC/Kodi process.
#
DPRIORITY=no

# Disable overclocking (and overvolting) when the screensaver is started.
# When the screensaver is stopped, it will enable overclocking again.
# It does so by changing the CPUFreq governor to "conservative", which doesn't
# actually disable overclocking, but tries to set it to the minimum required
# by current system load.
#
SCR_OVERCLOCK_OFF="yes"
```

mkreisl, 2019-02-26:
@florian-asche Thanks for posting /etc/default/xbmc
Unfortunately the more important file is missing.
On my system, the file is named `rpi_2708_1001.xml`, but on a fresh install, it seems to be `rpi_2708_1001_CEC_Adapter.xml`. I do not know which logic is behind this naming scheme. 
Anyway, please post this file, this file holds the CEC settings

florian-asche, 2019-02-26:
Here is the content from the file:
```
<settings>
    <setting id="activate_source" value="1" />
    <setting id="button_release_delay_ms" value="0" />
    <setting id="button_repeat_rate_ms" value="0" />
    <setting id="cec_hdmi_port" value="1" />
    <setting id="cec_standby_screensaver" value="0" />
    <setting id="cec_wake_screensaver" value="1" />
    <setting id="connected_device" value="36037" />
    <setting id="device_name" value="Kodi" />
    <setting id="device_type" value="1" />
    <setting id="double_tap_timeout_ms" value="300" />
    <setting id="enabled" value="1" />
    <setting id="pause_or_stop_playback_on_deactivate" value="36045" />
    <setting id="pause_playback_on_deactivate" value="0" />
    <setting id="physical_address" value="0" />
    <setting id="power_avr_on_as" value="0" />
    <setting id="send_inactive_source" value="1" />
    <setting id="standby_devices" value="36037" />
    <setting id="standby_devices_advanced" value="" />
    <setting id="standby_pc_on_tv_standby" value="36028" />
    <setting id="standby_tv_on_pc_standby" value="1" />
    <setting id="tv_vendor" value="0" />
    <setting id="use_tv_menu_language" value="0" />
    <setting id="wake_devices" value="36037" />
    <setting id="wake_devices_advanced" value="" />
</settings>

```

mkreisl, 2019-02-27:
@florian-asche 
> setting id="cec_wake_screensaver" value="1" /

This is the setting which is responsible for turning on TV when screensaver stops. I can't tell you why screensaver stops (maybe installed addon is culprit, or any events are displayed via toast-message or access via JSON or web interface).

Anyway, setting this to `value=0` will stop powering TV on after screensaver goes inactive. You can change this setting in Kodi->Settings->System->Input->Peripherals->CEC Adapter or edit file manually **after** stopping Kodi