xbianonpi/xbian #865: Wrong font name on estouchy theme causes subtitles to not show up

anselal, 2019-02-20:
Hi there,

this is more of an info rather than an issue. After I upgraded xbian to `20190215-0` and at the same time kodi to `18.1` I noticed that my subtitles disappeared.

After a quick search I found that the fonts are set in a `Font.xml` file so I searched my filesystem and found 2 `Font.xml` files inside the following folders:

- /usr/local/share/kodi/addons/skin.estouchy/xml/Font.xml
- /usr/local/share/kodi/addons/skin.estuary/xml/Font.xml

I also found the aria.ttf file inside `/usr/local/share/kodi/media/Fonts/arial.ttf`

I edited `/usr/local/share/kodi/addons/skin.estuary/xml/Font.xml` and saw that the font name of the arial.ttf file was written with a capital A like `Arial.ttf`. I replaced `Arial.ttf` with `arial.ttf`, rebooted and the subtitles appeared again !!!!

mkreisl, 2019-02-20:
@anselal Thanks for the hint. Could it be that you reported the wrong file?

`/usr/local/share/kodi/addons/skin.estuary/xml/Font.xml` has no **Arial.ttf** inside, only arial.ttf are there, but in the file `/usr/local/share/kodi/addons/skin.estouchy/xml/Font.xml` there are **Arial.ttf** names only.

So, I suppose you're using Estouchy skin, as written in header, right?

anselal, 2019-02-20:
Yeah sorry for that. I will edit my previous comment.

thnx

mkreisl, 2019-02-22:
I applied patch to `/usr/local/share/kodi/addons/skin.estouchy/xml/Font.xml`
With next build of `xbian-package-xbmc` this issue should be fixed

anselal, 2019-02-22:
Cool, thnx !!!

Unfortunately the fix was not permanent since the subtitles vanished again.

There seems to be a general issue with the subtitles and kodi v18.1, look at https://forum.kodi.tv/showthread.php?tid=337929

anselal, 2019-02-28:
@mkreisl any idea how to fix this ?

mkreisl, 2019-03-01:
@anselal how to fix what?
I wrote that it has been fixed in [sources](https://github.com/xbianonpi/xbian-sources-xbmc/commit/1895d6a2dfa3acc353c92e3558a3781c19a85068) already, and will be fixed in package after next build run. This will be happen when 18.2 becomes stable.
If you don't want do wait, 18.2-RC package are available in staging repository containing that fix already

anselal, 2019-03-01:
@mkreisl besides the thing with the theme there seems to be an issue with 18.1 in general, see 
https://forum.kodi.tv/showthread.php?tid=337929

mkreisl, 2019-03-01:
Shield == Android based, but we do not have Android

mkreisl, 2019-04-23:
Kodi 18.2 final has been built tonight. So we can close this issue now

closed 2019-04-23 by mkreisl