xbianonpi/xbian #866: xbian-backup.sh under cron not getting correct DWM value

jurgenvanimpe, 2019-02-22:


mkreisl, 2019-02-22:
@jurgenvanimpe A bit more text would be helpful

There is no "xbian-backup.sh"

Which value would be the right one and which value is set for DWM?

mkreisl, 2019-02-22:
Hmmm, calling `/etc/{cron.daily,cron.weekly,cron.monthly}/xbian-backup` **was never the idea to ba called by another script**, it's invoked by `(ana)cron`, that's why it is stored there

jurgenvanimpe, 2019-02-22:
I was trying to get the anacron going but with no luck, so I tried to call the script directly myself for testing purposes. Might be that it works perfectly when called from anacron. But I'm not getting that one to work.
I read all your old posts from 2014 about it.

mkreisl, 2019-02-22:
cat /etc/cron.daily/xb
```
#!/bin/sh

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

. /etc/default/xbian-snap

[ "$ENABLEDSCHEDULE" = yes ] || exit 0

DWM="${0##*.}"; DWM="${DWM%%/*}"

echo "DWM=$DWM"
```

If I run the testscript manually by full path, it gives me correct output
```
root@kmxbilr2 ~ # /etc/cron.daily/xb
DWM=daily
root@kmxbilr2 ~ # 
```

jurgenvanimpe, 2019-02-22:
Thanks for the fast reply.
I adjusted the file like you did. This is my cat output:
`sudo head -n 12 /etc/cron.daily/xbian-backup`

```
#!/bin/sh

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

. /etc/default/xbian-snap

[ "$ENABLEDSCHEDULE" = yes ] || exit 0

DWM="${0##*.}"; DWM="${DWM%%/*}"

echo "DWM=$DWM"

```
Result of runnin `cd /etc/cron.daily && sudo ./xbian-backup`:
`DWM=`
But strangely when I run: `sudo sh /etc/cron.daily/xbian-backup` I get:
`DWM=daily
....Starts running ....`

I can't explain this. I didn't study IT or anything and I'm trying to figure it out.
Hope you can understand it.

mkreisl, 2019-02-22:
I can understand this, but again, it's not an issue. 
The script **requires** the full path to figure out from which path (means which cron job is currently running) it has been called

jurgenvanimpe, 2019-02-22:
Thanks for clarification. Then this is not an issue.
I'll turn my direction to the forum if I have questions on how it exactly works.
Because I'm still looking on how to set the timing when it will perform the backup and to get it running properly.
Sorry for the inconvience.

closed 2019-02-22 by jurgenvanimpe

mkreisl, 2019-02-22:
The script is called by anacron per default. If you want to get the time when the jobs are started, please look into file `/etc/anacrontab` and file `/etc/cron.d/anacron`
In the first one is the delay in minutes configured for the daily, weekly and monthy jobs
In the second one is the start time of anacron configured

mk01, 2019-02-26:
one can use readlink -f on $0 first like this:
```
FP="$(readlink -f ${0})"
DWM="${FP##*.}"; DWM="${DWM%%/*}"
```

mkreisl, 2019-02-26:
Hi, @mk01 still alive :smile:

> one can use readlink -f on $0 first like this:
> 
> ```
> FP="$(readlink -f ${0})"
> DWM="${FP##*.}"; DWM="${DWM%%/*}"
> ```

This does not work if called from another link, for example /etc/cron.monthl/xbian-backup returns daily instead of monthly:

```
root@kmxbilr2 ~ # cat /etc/cron.daily/xb
#!/bin/sh

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

. /etc/default/xbian-snap

[ "$ENABLEDSCHEDULE" = yes ] || exit 0

FP="$(readlink -f ${0})"; DWM="${FP##*.}"; DWM="${DWM%%/*}"

echo "DWM=$DWM"
root@kmxbilr2 ~ # ls -la /etc/cron.monthly/xb
lrwxrwxrwx 1 root root 16 Feb 26 14:35 /etc/cron.monthly/xb -> ../cron.daily/xb
root@kmxbilr2 ~ # /etc/cron.monthly/xb
DWM=daily
root@kmxbilr2 ~ #
```

realpath -s $0 does it however:

```
root@kmxbilr2 /etc/cron.monthly # cat /etc/cron.daily/xb
#!/bin/sh

PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

. /etc/default/xbian-snap

[ "$ENABLEDSCHEDULE" = yes ] || exit 0

echo $0
FP="$(realpath -s $0)"; DWM="${FP##*.}"; DWM="${DWM%%/*}"

echo "DWM=$DWM"
root@kmxbilr2 /etc/cron.monthly # ls -la xb
lrwxrwxrwx 1 root root 16 Feb 26 14:35 xb -> ../cron.daily/xb
root@kmxbilr2 /etc/cron.monthly # ./xb
./xb
DWM=monthly
```

jurgenvanimpe, 2019-02-27:
Adjusting the files `/etc/cron.daily/anacron` and `/etc/anacrontab` works flawless now. I renamed also the script to make sure it starts second in row. I also added a line in the end of the script to trigger rsync to copy the backup off-site. Thanks!

mkreisl, 2019-02-27:
@jurgenvanimpe, modifying `/etc/cron.daily/xbian-backup` script and renaming it is not a good idea, because after next update of the package owning this file (in this case its `xbian-package-config-shell`), you will have 2 scripts making backups. And, you break links in `/etc/cron.weekly` and `/etc/cron.monthly`

jurgenvanimpe, 2019-02-27:
@mkreisl The links from cron.monthly/daily were already updated. Getting 2 files that would start making backups is something I didn't forsee. Do you have an elegant suggestion on how to get more control over the timing on execution of `xbian-backup` without renaming?
And the line at the end that I added to perform an rsync, would this be deleted after an update if the `xbian-backup` gets overwritten?
Thought I found a great solution.... appears not to :)

mkreisl, 2019-02-27:
@jurgenvanimpe 
> Do you have an elegant suggestion on how to get more control over the timing on execution of `xbian-backup` without renaming?

No

You could do this by bypassing the original logic:

1) rename the script (you already did this)
2) ignore ENABLEDSCHEDULE variable in your script (remove this line)
3) set ENABLEDSCHEDULE=no in /etc/default/xbian-snap

Be happy :smile: 

> And the line at the end that I added to perform an rsync, would this be deleted after an update if the `xbian-backup` gets overwritten?

If you rename the script, this file is save against updates