xbianonpi/xbian #868: hid_aureal module for rpi2

knedlyk, 2019-04-22:
Please add hid_aureal module for rpi2 kernel package. It exists for rpi, but but not for rpi2. 
I didn't find in repositories xbian kernel headers packages for rpi2, why these packages are missed?

mkreisl, 2019-04-22:
> Please add hid_aureal module for rpi2 kernel package. It exists for rpi, but but not for rpi2.

Will be done tomorrow

> I didn't find in repositories xbian kernel headers packages for rpi2, why these packages are missed?

Packages `linux-headers-bcm2836` and `linux-libc-dev` are available as usual

knedlyk, 2019-04-22:
Ach, ok, different name of the package. Thanks for the tip, now I can compile module by myself. But it would be good to have it in the official package.

knedlyk, 2019-04-22:
hid-aureal should be load before usbhid, but it is impossible because usbhid module is built in kernel. It should to be compiled as a module <M>, not as <*>.

mkreisl, 2019-04-22:
usbhid was **never** built as module

knedlyk, 2019-04-22:
Yes, my mistake, sorry, I meant hid-generic, CONFIG_HID_GENERIC=m as module.

knedlyk, 2019-04-23:
Thanks, my remote is working!

closed 2019-04-23 by knedlyk