xbianonpi/xbian #869: Upgrade package python-ldb_2%3a1.1.27-1+deb9u1_armhf.deb failed

anselal, 2019-05-03:
Preparing to unpack .../python-ldb_2%3a1.1.27-1+deb9u1_armhf.deb ...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: warning: subprocess old pre-removal script returned error exit status 1
dpkg: trying script from the new package instead ...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: error processing archive /var/cache/apt/archives/python-ldb_2%3a1.1.27-1+deb9u1_armhf.deb (--unpack):
 subprocess new pre-removal script returned error exit status 1
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
  File "/usr/share/python/debpython/version.py", line 24, in <module>
    from ConfigParser import SafeConfigParser
ImportError: No module named 'ConfigParser'
dpkg: error while cleaning up:
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/python-ldb_2%3a1.1.27-1+deb9u1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

mkreisl, 2019-05-03:
I suppose either your file system is broken, disk is full or download of this package failed.

Please check all of them

I do have that file installed, no problem seen:

```
systemd (232-25+deb9u11) wird eingerichtet ...
addgroup: Die Gruppe »systemd-journal« existiert bereits als Systemgruppe. Programmende.
(Lese Datenbank ... 72560 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../libldb1_2%3a1.1.27-1+deb9u1_armhf.deb ...
Entpacken von libldb1:armhf (2:1.1.27-1+deb9u1) über (2:1.1.27-1+b1) ...
libldb1:armhf (2:1.1.27-1+deb9u1) wird eingerichtet ...
(Lese Datenbank ... 72559 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../python-ldb_2%3a1.1.27-1+deb9u1_armhf.deb ...
Entpacken von python-ldb (2:1.1.27-1+deb9u1) über (2:1.1.27-1+b1) ...
python-ldb (2:1.1.27-1+deb9u1) wird eingerichtet ...
(Lese Datenbank ... 72558 Dateien und Verzeichnisse sind derzeit installiert.)
```

anselal, 2019-05-03:
The disk is not full and the internet connection is up.
I'll check my filesystem and come back.

anselal, 2019-05-03:
The disk is fine as well.

According to the error python can't interpret the new style of the exceptions.
changing default python version from 3 to 2 seemed to resolve the problem.

don't know why :(

closed 2019-11-20 by f1vefour