xbianonpi/xbian #870: Release for Raspberry Pi 4

peci1, 2019-08-30:
As I understand right now, Raspberry Pi 4 is not yet supported, because it needs Debian Buster as the base OS. 

I know people on forum already asked this, but I wanted some easily trackable place where people will know when it's ready.

Is somebody currently working on this? I've seen some buster development tracks here on github, but they don't seem to be building their artifacts to the APT repo yet...

mkreisl, 2019-08-31:
> As I understand right now, Raspberry Pi 4 is not yet supported, because it needs Debian Buster as the base OS.

That's not correct. Debian Buster ist not required. The main problem is that Kodi requires a completely rework to support Pi4 as it should. Actually, the existing solution (LE) is just a workaround.

> I know people on forum already asked this, but I wanted some easily trackable place where people will know when it's ready.

It's impossible to say when Kodi supports Pi4 completely. maybe with Kodi 19, but its completely unsure

> Is somebody currently working on this? I've seen some buster development tracks here on github, but they don't seem to be building their artifacts to the APT repo yet...

All work for the move from Stretch to Buster has been done already. Needs a bit more testing, but in one ore two weeks images will be built with Debian Buster

peci1, 2019-08-31:
Thanks for the info, @mkreisl . Would you please leave here a notice when the Buster images start being built?

peci1, 2019-09-13:
So this image https://sourceforge.net/projects/xbian/files/release/XBian_2019.09.13_rpi3.img.gz/download should already be based on buster? I'll give it a try later on my Raspberry pi 4 and see what happens :)

mkreisl, 2019-09-13:
> So this image https://sourceforge.net/projects/xbian/files/release/XBian_2019.09.13_rpi3.img.gz/download should already be based on buster?

Yes, it does

mkreisl, 2019-09-13:
But of course, it DOES NOT WORK with Pi4

peci1, 2019-09-13:
Hmm, what's the blocker? The kernel in buster should be ready for Pi4, shouldn't it? And if I'm not wrong, the CPU architecture should be the same as Rpi3b+...

mkreisl, 2019-09-13:
You are wrong. SOC is completely different

1) Firmware (in all images) does not support Pi4
2) XBian kernel (in all images) does not support Pi4
3) Kodi does not run under Pi4

I'm running XBian on my local Pi4 with devel kernel (5.2.14) and  latest firmware from RPF

peci1, 2019-09-13:
Right now I have a stable Rasbian Buster running on Pi4 with Kodi working. So it'd seem at least kernel and Kodi should be ok. But I don't know if Xbian uses the same kernels Raspbian does (from your post it seems it doesn't).

joystickj, 2019-10-05:
Und jetzt?

joystickj, 2019-10-08:
Nice! Downloading it now!

npmstart-pray, 2019-11-04:
from the radio silence can I assume it's working?

CurlyMoo, 2019-11-04:
Here, the sound output doesn't work.

npmstart-pray, 2019-11-04:
audio over hdmi or ???

CurlyMoo, 2019-11-04:
On both HDMI ports

peci1, 2019-11-04:
Yes, it works for me. The HDMI audio works only on the first HDMI port.

npmstart-pray, 2019-11-04:
I think I'll give it a go, then.

npmstart-pray, 2020-02-25:
the current img rpi4-2020-02-21 is broken - boots straight to recovery

mkreisl, 2020-02-26:
> the current img rpi4-2020-02-21 is broken - boots straight to recovery

Yes, indeed. Pi4 does not boot, Pi3B+ does. Must have something to do with kernel 5.4

Pi4 does not find sd card, no idea why. In case of booting via PINN (which is usually my test environment), everything is ok though.

npmstart-pray, 2020-02-26:
> Latest image(s) and Kernel 5.4.20 package has been 
> deleted

going back to 4.x.x may be a tad defeatist:

> Must have something to do with kernel 
> 5.4

this latest, 5.4.20+, I agree.  the one before, 5.4.19+, was brilliant _for me_.

mkreisl, 2020-02-26:
Hmmm, no idea what happened between .19 and .20.

Anyway, I modified kernel config, build new 5.4.20 kernel and now Pi4 is booting.I think it is fixed now and  new kernel 5.4.22 will be available in 2 hours

npmstart-pray, 2020-03-09:
The version of kodi (18.6) included with update to kernel 5.4.24 crashes, and kicks me out to a command line

mkreisl, 2020-03-09:
Does prev version of kodi work (18.5)?

npmstart-pray, 2020-03-09:
Yes. I rolled back and OS updates make that version work better. 
(I was about to report this...)

Sent from nowhere near my ready room with my new tricorder

mkreisl, 2020-03-09:
Oh yes, I see. In xbian-package-xbmc 18.6 binaries for RPi4 has not been included

Sorry for that, will fix it asap

npmstart-pray, 2020-03-10:
Will report back once I’ve tried it

Sent from nowhere near my ready room with my new tricorder

npmstart-pray, 2020-10-06:
kodi v19 alpha2 showed up in my feed as pushed...